mongoose-os-apps/shelly-homekit #923: Investigate Wireless Accessory Configuration Protocol

d3vi1, 2022-02-01:
The WiFi setup experience for HomeKit devices could be a lot better if Wireless Accessory Configuration protocol is used. As a protocol, WAC depends on a special WiFi IE when in AP mode (0x00A04000) and the following TLVs:

1. Flags (bit array)
2. UTF-8 Device Name
3. UTF-8 Device Manufacturer
4. UTF-8 Model
5. 3 byte unknown
6. 6 byte MAC Address

The known bit array flags are:

- `0x4000` - Unconfigured state
- `0x2000` - Always TRUE
- `0x1000` - Wake on Wireless is supported
- `0x0200` - Wifi Protected Setup is supported
- `0x0100` - WiFi Protected Setup is active
- `0x0002` - 2.4GHz Support
- `0x0001` - 5GHz Support

If an iDevice sees such a wireless AP, it will join it and look for the _mfi-config._tcp.local mDNS announcement.

The device will then expose 3 HTTP endpoints:
- `/auth-setup`
- `/config`
- `/configured`

This should create a seamless experience for the Shelly HomeKit devices. I will do my best to create a PCAP of a WAC joining and configuration if there is an interest in this.

There is some documentation already available on "[Telit WAC Application Note](https://y1cj3stn5fbwhv73k0ipk1eg-wpengine.netdna-ssl.com/wp-content/uploads/2018/02/GS2K-WAC-Provisioning-ADK-Application-Note_r1.0.pdf)" and the "[Silicon Labs WAC User Guide](https://www.silabs.com/documents/login/user-guides/WAC_User_Guide.pdf)"

The actual mfi-config service seems to be included in the `HAPIPServiceDiscoverySetMFiConfigService` part of HAPIPServiceDiscovery.c in the HomeKitADK if `HAPIPServiceDiscoveryType` is `kHAPIPServiceDiscoveryType_MFiConfig`.

It is my understanding that for an unconfigured device, if you use WAC you should use `kHAPIPServiceDiscoveryType_MFiConfig`. When the device joins your wireless network, it should then switch to `kHAPIPServiceDiscoveryType_HAP`

timoschilling, 2022-02-01:
Interesting!

But I see tree points:
1. Maybe the right repository for this is https://github.com/mongoose-os-libs/homekit-adk, @rojer what do you think? If yes you can simply use "Transfer issue" in the right sidebar.
2. Is that really a benefit? I ask because the device must be already in the wifi to flash this firmware.
3. Does it work with uncertified / not MFi devices?

d3vi1, 2022-02-02:
> 1. Maybe the right repository for this is https://github.com/mongoose-os-libs/homekit-adk, @rojer what do you think? If yes you can simply use "Transfer issue" in the right sidebar.

The Mongoose Homekit-adk might be a better repository, but I can't transfer the issue myself (no "Transfer issue" button). And it would also require Mongoose WiFi API to expose a wrapper of `esp_wifi_set_vendor_ie`.

> 2. Is that really a benefit? I ask because the device must be already in the wifi to flash this firmware.

Sure there is. Once you convert the device to the HomeKit firmware, you might have to add it one or more times to the network. Imagine changing the SSID & resetting to factory in order to reuse the device in the same or another instance.

> 3. Does it work with uncertified / not MFi devices?

The initial part of WAC is the WiFi IEs, that has nothing to do with MFi. It's worth asking on the Apple HomeKit-ADK if the `kHAPIPServiceDiscoveryType_MFiConfig` can be used without the MFi chip, or if the `kHAPIPServiceDiscoveryType_HAP` is the only one available.

timoschilling, 2022-02-02:
sorry I didn't express myself clearly enough, the hint about moving was for @rojer only he can do that. But maybe he can't do that as well, because homekit-adk is in a different organisation.

rojer, 2022-02-02:
WAC is a non-started for us because... Apple, essentially.
WAC requires an official crypto token to authenticate the accessory before any credential exchange happens.
so unfortunately, this is never going to happen for us.

closed 2022-02-02 by rojer