mongoose-os-apps/shelly-homekit #924: Help with homekit

Samcam1228, 2022-02-02:
I am wondering if there is a way to use a Shelly 1 as a sensor instead of a switch in home kit. I have a Shelly 1 hooked up to a reed switch and I would like it to send a notification if the switch changes state using home kit. I would prefer that it shows up as a notification, not a text message. When I try adding it with home kit it only allows it to be added as a switch, light or fan so I can not create an automation to send an alert

timoschilling, 2022-02-02:
Open the web ui of the shelly in your browser, there you can change the "Input Mode" to "Detached" and then you can set the "HAP Type" to "Contact Sensor".
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Switch-Settings
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Input-Switch-Settings#motion-sensor--occupancy-sensor--contact-sensor--doorbell

Samcam1228, 2022-02-02:
That it! Thank you. Is there a way to change the label for the “contact sensor state” in home kit? I was hoping to be able to change it to something different than open close for say a flood alarm

timoschilling, 2022-02-02:
You can rename the sensor via long tap on the Icon.

PR #929 (2022-03-02): implement leak sensor
closed #924
closed 2022-03-02 by rojer