mongoose-os-apps/shelly-homekit #928: Unable to update Shelly 2.5

fparodo, 2022-02-09:
Hello,

I try to update my shelly 2.5 from   20210311-193713/2.8.0-g1e8cd09-master to the new version but dosen't work 
From  command in the terminal  I receive this error   ./flash-shelly.py --force finestasalone.local -V 2.9.0
Looking for Shelly devices...
Host: finestasalone.local is not a Shelly device.

and  from UI   "Error uploading: TypeError: Failed to fetch" 
Below the log 

894603 shelly_main.cpp:376     HAP server state: 1
907771 mgos_ota_core.c:1329    UID: 03f54674db8763b9, license: none
918684 mgos_mdns.c:82          Listening on udp://:5353
936102 mgos_mongoose.c:66      New heap free LWM: 22560
945848 mgos_mdns.c:55          mDNS socket closed
955431 mgos_mdns.c:82          Listening on udp://:5353
964727 mgos_net.c:86           WiFi STA: connecting
976964 mgos_mdns.c:55          mDNS socket closed
987464 mgos_mdns.c:82          Listening on udp://:5353
3630566 esp_main.c:137          SDK: scandone
3640018 esp_main.c:137          SDK: state: 0 -> 2 (b0)
3652545 esp_main.c:137          SDK: state: 2 -> 3 (0)
3664657 esp_main.c:137          SDK: state: 3 -> 5 (10)
3673147 esp_main.c:137          SDK: add 0
3680549 esp_main.c:137          SDK: aid 8
3688164 esp_main.c:137          SDK: cnt 
3708485 esp_main.c:137          SDK: 
3717741 esp_main.c:137          SDK: connected with Air, channel 1
3731529 esp_main.c:137          SDK: ip:192.168.1.20,mask:255.255.255.0,gw:192.168.1.1
3746176 mgos_wifi.c:134         WiFi STA: Connected, BSSID 00:24:36:ad:ae:bb ch 1 RSSI -69
3756347 mgos_net.c:90           WiFi STA: connected
3808027 mgos_net.c:101          WiFi STA: ready, IP 192.168.1.20, GW 192.168.1.1, DNS 8.8.8.8
3825238 mgos_mongoose.c:66      New heap free LWM: 22064
3853864 mgos_http_server.c:180  0x3fff644c HTTP connection from 192.168.1.122:49612
3868035 mgos_mongoose.c:66      New heap free LWM: 21488
3921440 mgos_http_server.c:180  0x3fff661c HTTP connection from 192.168.1.122:49613
3934883 mgos_http_server.c:187  0x3fff661c GET /
3949287 mgos_mongoose.c:66      New heap free LWM: 21144
3962811 mgos_mongoose.c:66      New heap free LWM: 20384
4000231 mgos_mongoose.c:66      New heap free LWM: 19392
4038858 mgos_http_server.c:180  0x3fff644c HTTP connection from 192.168.1.122:49614
4056856 mgos_http_server.c:187  0x3fff661c GET /favicon.ico
4076679 mgos_mongoose.c:66      New heap free LWM: 19056
4132491 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.122:49614
4151549 mgos_mongoose.c:66      New heap free LWM: 19048
4211291 HAPPlatformTCPStrea:121 0x3fff66d4 192.168.1.214:53855 Incoming HAP connection, ns 0/0/12
4226402 HAPPlatformTCPStrea:276 0x3fff66d4 192.168.1.214:53855 Accepted HAP connection, ns 0/1/12 ts 0x3fff6f94
5357444 HAPPlatformTCPStrea:105 0x3fff66d4 192.168.1.214:53855 HAP connection closed, ns 0/1/12 i 2
5369257 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6f94 nc 0x0 0/0/12
5385975 mgos_sntp.c:98          SNTP query to time.google.com
5395086 mgos_mongoose.c:66      New heap free LWM: 18944
5410633 mgos_mongoose.c:66      New heap free LWM: 17128
5421833 mgos_mongoose.c:66      New heap free LWM: 17112
5443969 HAPPlatformTCPStrea:121 0x3fff67e4 192.168.1.197:49849 Incoming HAP connection, ns 0/0/12
5544189 mgos_sntp.c:60          SNTP reply from 216.239.35.12: time 1644415409.194953, local 5.531074, delta 1644415403.663879
6211427:1644415409857851 HAPPlatformTCPStrea:121 0x3fff686c 192.168.1.214:53856 Incoming HAP connection, ns 1/0/12
7286332 HAPPlatformTCPStrea:276 0x3fff67e4 192.168.1.197:49849 Accepted HAP connection, ns 1/1/12 ts 0x3fff6bf4
8772014 shelly_main.cpp:554     Up 8.76, HAP 1/1/12 ns 1, RAM: 19476/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
10353786 HAPPlatformTCPStrea:276 0x3fff686c 192.168.1.214:53856 Accepted HAP connection, ns 0/2/12 ts 0x3fff6a94
11520673 HAPPlatformTCPStrea:105 0x3fff686c 192.168.1.214:53856 HAP connection closed, ns 0/2/12 i 2
11532512 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6a94 nc 0x0 0/1/12
13689255 esp_main.c:137          SDK: pm open,type:0 0
16771308 shelly_main.cpp:554     Up 16.76, HAP 0/1/12 ns 1, RAM: 20112/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
19618151 HAPPlatformTCPStrea:121 0x3fff6c1c 192.168.1.214:53857 Incoming HAP connection, ns 0/1/12
19635804 HAPPlatformTCPStrea:276 0x3fff6c1c 192.168.1.214:53857 Accepted HAP connection, ns 0/2/12 ts 0x3fff6a94
23187080 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.122:49614
24181186 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.122:49614
25182335 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.122:49614
25859297 shelly_main.cpp:554     Up 25.85, HAP 0/2/12 ns 2, RAM: 21228/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
26179586 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.122:49614
[... 39 lines cut ...]
53104715 shelly_main.cpp:341     === Starting HAP server (CN 1)
53173087 HAPPlatformTCPStrea:187 Listening on 9001
53195327 shelly_main.cpp:376     HAP server state: 1
53235845 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.122:49625
53417807 HAPPlatformTCPStrea:121 0x3fff64d4 192.168.1.214:53858 Incoming HAP connection, ns 0/0/12
53432042 HAPPlatformTCPStrea:276 0x3fff64d4 192.168.1.214:53858 Accepted HAP connection, ns 0/1/12 ts 0x3fff5bbc
54620055 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.122:49625
55238866 mg_rpc.c:305            Shelly.GetInfo via WS_in 192.168.1.122:49625
58962858 shelly_main.cpp:554     Up 58.95, HAP 0/1/12 ns 1, RAM: 22068/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
66093312 HAPPlatformTCPStrea:121 0x3fff68dc 192.168.1.195:50825 Incoming HAP connection, ns 0/1/12
66107563 HAPPlatformTCPStrea:276 0x3fff68dc 192.168.1.195:50825 Accepted HAP connection, ns 0/2/12 ts 0x3fff6f34
67291096 shelly_main.cpp:554     Up 67.28, HAP 0/2/12 ns 2, RAM: 18364/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
67777282 mgos_http_server.c:180  0x3fff6964 HTTP connection from 192.168.1.122:49631
67899474 shelly_debug.cpp:203    Tailing
74962683 shelly_main.cpp:554     Up 74.95, HAP 0/2/12 ns 2, RAM: 20484/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
82962561 shelly_main.cpp:554     Up 82.95, HAP 0/2/12 ns 2, RAM: 20484/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
90962781 shelly_main.cpp:554     Up 90.95, HAP 0/2/12 ns 2, RAM: 20484/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
98962478 shelly_main.cpp:554     Up 98.95, HAP 0/2/12 ns 2, RAM: 20484/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
106962512 shelly_main.cpp:554     Up 106.95, HAP 0/2/12 ns 2, RAM: 20484/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
111142489 HAPPlatformTCPStrea:121 0x3fff7174 192.168.1.197:49850 Incoming HAP connection, ns 0/2/12
111158162 HAPPlatformTCPStrea:276 0x3fff7174 192.168.1.197:49850 Accepted HAP connection, ns 0/3/12 ts 0x3fff655c
114962572 shelly_main.cpp:554     Up 114.95, HAP 0/3/12 ns 3, RAM: 19968/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0

Greetings 
Francesco.

andyblac, 2022-02-09:
`53051828 mgos_ota_core.c:811 Update finished, result -2 (Invalid checksum)`

download failed, seems like a local internet issue. plus 2.9.0 is quite now, yo update to latest version just run `./flash-shelly.py finestasalone.local`

fparodo, 2022-02-09:
Not work 

 ./flash-shelly.py finestasalone.local
Looking for Shelly devices...

Host: finestasalone.local is not a Shelly device.

Devices found: 0 Upgradeable: 0 Flashed: 0

Log: 

578962453 shelly_main.cpp:554     Up 578.95, HAP 0/3/12 ns 3, RAM: 20508/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
586962486 shelly_main.cpp:554     Up 586.95, HAP 0/3/12 ns 3, RAM: 20508/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
592023121 mgos_http_server.c:180  0x3fff7414 HTTP connection from 192.168.1.122:49882
592081395 mgos_http_server.c:180  0x3fff7284 HTTP connection from 192.168.1.122:49883
592093758 mgos_http_server.c:187  0x3fff7284 GET /settings
592133091 mgos_http_server.c:180  0x3fff7284 HTTP connection from 192.168.1.122:49884
592147151 mg_rpc.c:285            No handler for Shelly.GetInfoExt
594962304 shelly_main.cpp:554     Up 594.95, HAP 0/3/12 ns 3, RAM: 20516/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
602962377 shelly_main.cpp:554     Up 602.95, HAP 0/3/12 ns 3, RAM: 20704/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
603597494 mgos_http_server.c:180  0x3fff733c HTTP connection from 192.168.1.122:49887
603619960 mgos_http_server.c:180  0x3fff57ac HTTP connection from 192.168.1.122:49888
603631365 mgos_http_server.c:187  0x3fff57ac GET /settings
603673456 mgos_http_server.c:180  0x3fff55cc HTTP connection from 192.168.1.122:49889
603724045 mg_rpc.c:285            No handler for Shelly.GetInfoExt
609885747 mgos_http_server.c:180  0x3fff7284 HTTP connection from 192.168.1.122:49895
609906541 mgos_http_server.c:180  0x3fff55cc HTTP connection from 192.168.1.122:49896
609918905 mgos_http_server.c:187  0x3fff55cc GET /settings
609966990 mgos_http_server.c:180  0x3fff7414 HTTP connection from 192.168.1.122:49897
609981135 mg_rpc.c:285            No handler for Shelly.GetInfoExt
610962588 shelly_main.cpp:554     Up 610.95, HAP 0/3/12 ns 3, RAM: 20328/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0

andyblac, 2022-02-09:
please give me debug log `./flash-shelly.py finestasalone.local --log-file debug.log` it should create a file called 'debug.log' please post it here

andyblac, 2022-02-09:
also try without host name, allow the script to discover your device.

fparodo, 2022-02-09:
Log: 

2022-02-09 15:17:58,078 TRACE 913 load_security
2022-02-09 15:17:58,079 DEBUG 1046 OS: Darwin
2022-02-09 15:17:58,079 DEBUG 1047 app_version: 3.0.0
2022-02-09 15:17:58,079 DEBUG 1048 do_all: False
2022-02-09 15:17:58,084 DEBUG 1049 dry_run: False
2022-02-09 15:17:58,084 DEBUG 1050 quiet_run: False
2022-02-09 15:17:58,085 DEBUG 1051 silent_run: False
2022-02-09 15:17:58,085 DEBUG 1052 timeout: 20
2022-02-09 15:17:58,085 DEBUG 1053 reboot: False
2022-02-09 15:17:58,085 DEBUG 1054 manual_hosts: ['finestasalone.local'] (1)
2022-02-09 15:17:58,085 DEBUG 1055 user: admin
2022-02-09 15:17:58,085 DEBUG 1056 password: None
2022-02-09 15:17:58,085 DEBUG 1057 action: flash
2022-02-09 15:17:58,086 DEBUG 1058 mode: homekit
2022-02-09 15:17:58,086 DEBUG 1059 info_level: 3
2022-02-09 15:17:58,088 DEBUG 1060 fw_type_filter: all
2022-02-09 15:17:58,088 DEBUG 1061 model_type_filter: all
2022-02-09 15:17:58,088 DEBUG 1062 device_name_filter: all
2022-02-09 15:17:58,090 DEBUG 1063 version:
2022-02-09 15:17:58,091 DEBUG 1064 exclude:
2022-02-09 15:17:58,091 DEBUG 1065 local_file:
2022-02-09 15:17:58,091 DEBUG 1066 variant:
2022-02-09 15:17:58,091 DEBUG 1067 force: False
2022-02-09 15:17:58,091 DEBUG 1068 verbose: 5
2022-02-09 15:17:58,091 DEBUG 1069 hap_setup_code:
2022-02-09 15:17:58,091 DEBUG 1070 network_type:
2022-02-09 15:17:58,091 DEBUG 1071 ipv4_ip:
2022-02-09 15:17:58,092 DEBUG 1072 ipv4_mask:
2022-02-09 15:17:58,092 DEBUG 1073 ipv4_gw:
2022-02-09 15:17:58,092 DEBUG 1074 ipv4_dns:
2022-02-09 15:17:58,092 DEBUG 1075 log_filename: debug.log
2022-02-09 15:17:58,092 DEBUG 1658 [Manual Hosts]
2022-02-09 15:17:58,092 INFO 1659 Looking for Shelly devices...
2022-02-09 15:17:58,094 DEBUG 1661
2022-02-09 15:17:58,094 DEBUG 1662 [Manual Hosts] action finestasalone.local
2022-02-09 15:17:58,094 TRACE 921 get_security_data
2022-02-09 15:17:58,095 DEBUG 923 [Security] finestasalone.local
2022-02-09 15:17:58,095 DEBUG 932 [Security] username: admin from commandline
2022-02-09 15:17:58,095 DEBUG 933 [Security] password: None from commandline
2022-02-09 15:17:58,131 DEBUG 315 Device: finestasalone.local is Online
2022-02-09 15:17:58,204 TRACE 282 RESPONSE: <Response [404]>
2022-02-09 15:17:58,249 TRACE 282 RESPONSE: <Response [404]>
2022-02-09 15:17:58,249 INFO 293
2022-02-09 15:17:58,250 INFO 294 Host: finestasalone.local is not a Shelly device.
2022-02-09 15:17:58,250 INFO 1703
2022-02-09 15:17:58,250 INFO 1708 Devices found: 0 Upgradeable: 0 Flashed: 0
2022-02-09 15:17:58,251 INFO 1712 Log file created: debug.log

fparodo, 2022-02-09:
> also try without host name, allow the script to discover your device.

same error

andyblac, 2022-02-09:
give me new log without using hostname please.

andyblac, 2022-02-09:
the script does not see the host `finestasalone.loca` as a shelly device.

fparodo, 2022-02-09:
See below:

2022-02-09 15:20:53,046 DEBUG 244 [Device Scan] added: Finestramatr.local, IP address: 192.168.1.139, FW Type: homekit, Security Enabled: False to queue
2022-02-09 15:20:53,046 DEBUG 245
2022-02-09 15:20:53,049 DEBUG 227 [Device Scan] found device: finestasalone.local, IP address: 192.168.1.20
2022-02-09 15:20:53,049 TRACE 228 [Device Scan] info: ServiceInfo(type='_http._tcp.local.', name='finestasalone._http._tcp.local.', addresses=[b'\xc0\xa8\x01\x14'], port=80, weight=0, priority=0, server='finestasalone.local.', properties={b'id': b'shellyswitch25-73C352', b'arch': b'esp8266', b'app': b'switch25', b'fw_version': b'2.8.0', b'fw_id': b'20210311-
193713/2.8.0-g1e8cd09-master'}, interface_index=None)
2022-02-09 15:20:53,049 TRACE 229 [Device Scan] properties: {'id': 'shellyswitch25-73C352', 'arch': 'esp8266', 'app': 'switch25', 'fw_version': '2.8.0', 'fw_id': '20210311-193713/2
.8.0-g1e8cd09-master'}
2022-02-09 15:20:53,056 DEBUG 315 Device: finestasalone.local is Online
2022-02-09 15:20:53,123 TRACE 282 RESPONSE: <Response [404]>
2022-02-09 15:20:53,164 TRACE 282 RESPONSE: <Response [404]>
2022-02-09 15:20:53,164 INFO 293
2022-02-09 15:20:53,164 INFO 294 Host: finestasalone.local is not a Shelly device.
2022-02-09 15:20:53,165 DEBUG 245

andyblac, 2022-02-09:
are you using version 3 of the script?

andyblac, 2022-02-09:
please try this:

[flash-shelly.py.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/8033461/flash-shelly.py.zip)

fparodo, 2022-02-09:
i think it should be version 3.0 and is the latest update done by git I try your flash-shelly but I have the same error

andyblac, 2022-02-09:
ok, sorry try:
[flash-shelly.py.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/8033514/flash-shelly.py.zip)

fparodo, 2022-02-09:
Host: http://finestasalone.local
Device Name: finestasalone
Model: Shelly25
Firmware: HomeKit 2.8.0 ➜ 2.10.1
Do you wish to flash finestasalone to HomeKit firmware version 2.10.1 (y/n) ? y
Downloading Firmware...
Now Flashing HomeKit 2.10.1

flash-shelly version: 2.8.0
Try to update your script, maybe the bug is already fixed!
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/urllib3/response.py", line 438, in _error_catcher
    yield
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/urllib3/response.py", line 519, in read
    data = self._fp.read(amt) if not fp_closed else b""
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/http/client.py", line 458, in read
    n = self.readinto(b)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/http/client.py", line 502, in readinto
    n = self.fp.readinto(b)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/socket.py", line 669, in readinto
    return self._sock.recv_into(b)
ConnectionResetError: [Errno 54] Connection reset by peer

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests/models.py", line 760, in generate
    for chunk in self.raw.stream(chunk_size, decode_content=True):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/urllib3/response.py", line 576, in stream
    data = self.read(amt=amt, decode_content=decode_content)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/urllib3/response.py", line 541, in read
    raise IncompleteRead(self._fp_bytes_read, self.length_remaining)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/contextlib.py", line 131, in __exit__
    self.gen.throw(type, value, traceback)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/urllib3/response.py", line 455, in _error_catcher
    raise ProtocolError("Connection broken: %r" % e, e)
urllib3.exceptions.ProtocolError: ("Connection broken: ConnectionResetError(54, 'Connection reset by peer')", ConnectionResetError(54, 'Connection reset by peer'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./flash-shelly.py", line 941, in run_app
    self.device_scan()
  File "./flash-shelly.py", line 1579, in device_scan
    self.probe_device(device)
  File "./flash-shelly.py", line 1485, in probe_device
    self.parse_info(device_info, hk_flash_fw_version)
  File "./flash-shelly.py", line 1355, in parse_info
    self.write_flash(device_info)
  File "./flash-shelly.py", line 1105, in write_flash
    response = device_info.flash_firmware(revert)
  File "./flash-shelly.py", line 574, in flash_firmware
    response = requests.post(url=f'http://{self.wifi_ip}/update', auth=HTTPDigestAuth(self.username, self.password), files=files)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests/api.py", line 117, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests/sessions.py", line 529, in request
    resp = self.send(prep, **send_kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests/sessions.py", line 687, in send
    r.content
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests/models.py", line 838, in content
    self._content = b''.join(self.iter_content(CONTENT_CHUNK_SIZE)) or b''
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests/models.py", line 763, in generate
    raise ChunkedEncodingError(e)
requests.exceptions.ChunkedEncodingError: ("Connection broken: ConnectionResetError(54, 'Connection reset by peer')", ConnectionResetError(54, 'Connection reset by peer'))

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./flash-shelly.py", line 1584, in <module>
    sys.exit(main.run_app())
  File "./flash-shelly.py", line 947, in run_app
    traceback.trace_exception(exc_type, exc_value, exc_traceback, file=sys.stdout)
AttributeError: module 'traceback' has no attribute 'trace_exception'

Devices found: 8 Upgradeable: 1 Flashed: 0

andyblac, 2022-02-09:
@fparodo what does the device log say now ?

andyblac, 2022-02-09:
@fparodo can you try update your python to version 3.9.xx ?

fparodo, 2022-02-09:
ok, I try an now copy the last log

andyblac, 2022-02-09:
hmm 

```
2022-02-09 15:48:54,588 DEBUG 847 OS: Darwin
2022-02-09 15:48:54,588 DEBUG 848 app_version: 2.8.0
```
it should be 3.0.0 ???

fparodo, 2022-02-09:
.... sorry wrong debug.log copy, a test made with the old version

fparodo, 2022-02-09:
2022-02-09 16:01:39,046 TRACE 913 load_security
2022-02-09 16:01:39,046 DEBUG 1046 OS: Darwin
2022-02-09 16:01:39,046 DEBUG 1047 app_version: 3.0.0
2022-02-09 16:01:39,046 DEBUG 1048 do_all: False
2022-02-09 16:01:39,047 DEBUG 1049 dry_run: False
2022-02-09 16:01:39,047 DEBUG 1050 quiet_run: False
2022-02-09 16:01:39,048 DEBUG 1051 silent_run: False
2022-02-09 16:01:39,048 DEBUG 1052 timeout: 20
2022-02-09 16:01:39,048 DEBUG 1053 reboot: False
2022-02-09 16:01:39,048 DEBUG 1054 manual_hosts: ['finestasalone.local'] (1)
2022-02-09 16:01:39,048 DEBUG 1055 user: admin
2022-02-09 16:01:39,048 DEBUG 1056 password: None
2022-02-09 16:01:39,048 DEBUG 1057 action: flash
2022-02-09 16:01:39,048 DEBUG 1058 mode: homekit
2022-02-09 16:01:39,048 DEBUG 1059 info_level: 3
2022-02-09 16:01:39,048 DEBUG 1060 fw_type_filter: all
2022-02-09 16:01:39,048 DEBUG 1061 model_type_filter: all
2022-02-09 16:01:39,048 DEBUG 1062 device_name_filter: all
2022-02-09 16:01:39,048 DEBUG 1063 version:
2022-02-09 16:01:39,048 DEBUG 1064 exclude:
2022-02-09 16:01:39,048 DEBUG 1065 local_file:
2022-02-09 16:01:39,049 DEBUG 1066 variant:
2022-02-09 16:01:39,049 DEBUG 1067 force: False
2022-02-09 16:01:39,049 DEBUG 1068 verbose: 5
2022-02-09 16:01:39,049 DEBUG 1069 hap_setup_code:
2022-02-09 16:01:39,049 DEBUG 1070 network_type:
2022-02-09 16:01:39,049 DEBUG 1071 ipv4_ip:
2022-02-09 16:01:39,049 DEBUG 1072 ipv4_mask:
2022-02-09 16:01:39,049 DEBUG 1073 ipv4_gw:
2022-02-09 16:01:39,049 DEBUG 1074 ipv4_dns:
2022-02-09 16:01:39,049 DEBUG 1075 log_filename: debug.log
2022-02-09 16:01:39,049 DEBUG 1658 [Manual Hosts]
2022-02-09 16:01:39,049 INFO 1659 Looking for Shelly devices...
2022-02-09 16:01:39,049 DEBUG 1661
2022-02-09 16:01:39,049 DEBUG 1662 [Manual Hosts] action finestasalone.local
2022-02-09 16:01:39,050 TRACE 921 get_security_data
2022-02-09 16:01:39,052 DEBUG 923 [Security] finestasalone.local
2022-02-09 16:01:39,052 DEBUG 932 [Security] username: admin from commandline
2022-02-09 16:01:39,052 DEBUG 933 [Security] password: None from commandline
2022-02-09 16:01:39,074 DEBUG 315 Device: finestasalone.local is Online
2022-02-09 16:01:39,180 TRACE 282 RESPONSE: <Response [404]>
2022-02-09 16:01:39,260 TRACE 282 RESPONSE: <Response [404]>
2022-02-09 16:01:39,319 TRACE 282 RESPONSE: <Response [200]>
2022-02-09 16:01:39,365 TRACE 367 Device URL: http://192.168.1.20/rpc/Shelly.GetInfoExt
2022-02-09 16:01:39,366 TRACE 368 status code: 404
2022-02-09 16:01:39,366 TRACE 369 security: False
2022-02-09 16:01:39,367 DEBUG 396 Failed to lookup local information of finestasalone.local
2022-02-09 16:01:39,367 INFO 1703
2022-02-09 16:01:39,368 INFO 1708 Devices found: 0 Upgradeable: 0 Flashed: 0
2022-02-09 16:01:39,369 INFO 1712 Log file created: debug.log

andyblac, 2022-02-09:
ok try:
[flash-shelly.py.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/8033751/flash-shelly.py.zip)

fparodo, 2022-02-09:
2022-02-09 16:10:57,635 TRACE 916 load_security
2022-02-09 16:10:57,636 DEBUG 1049 OS: Darwin
2022-02-09 16:10:57,636 DEBUG 1050 app_version: 3.0.0
2022-02-09 16:10:57,636 DEBUG 1051 do_all: False
2022-02-09 16:10:57,636 DEBUG 1052 dry_run: False
2022-02-09 16:10:57,636 DEBUG 1053 quiet_run: False
2022-02-09 16:10:57,636 DEBUG 1054 silent_run: False
2022-02-09 16:10:57,636 DEBUG 1055 timeout: 20
2022-02-09 16:10:57,636 DEBUG 1056 reboot: False
2022-02-09 16:10:57,636 DEBUG 1057 manual_hosts:  (0)
2022-02-09 16:10:57,636 DEBUG 1058 user: admin
2022-02-09 16:10:57,636 DEBUG 1059 password: None
2022-02-09 16:10:57,636 DEBUG 1060 action: flash
2022-02-09 16:10:57,636 DEBUG 1061 mode: homekit
2022-02-09 16:10:57,637 DEBUG 1062 info_level: 3
2022-02-09 16:10:57,637 DEBUG 1063 fw_type_filter: all
2022-02-09 16:10:57,637 DEBUG 1064 model_type_filter: all
2022-02-09 16:10:57,637 DEBUG 1065 device_name_filter: all
2022-02-09 16:10:57,637 DEBUG 1066 version: 
2022-02-09 16:10:57,637 DEBUG 1067 exclude: 
2022-02-09 16:10:57,637 DEBUG 1068 local_file: 
2022-02-09 16:10:57,637 DEBUG 1069 variant: 
2022-02-09 16:10:57,637 DEBUG 1070 force: False
2022-02-09 16:10:57,637 DEBUG 1071 verbose: 5
2022-02-09 16:10:57,637 DEBUG 1072 hap_setup_code: 
2022-02-09 16:10:57,637 DEBUG 1073 network_type: 
2022-02-09 16:10:57,637 DEBUG 1074 ipv4_ip: 
2022-02-09 16:10:57,637 DEBUG 1075 ipv4_mask: 
2022-02-09 16:10:57,637 DEBUG 1076 ipv4_gw: 
2022-02-09 16:10:57,637 DEBUG 1077 ipv4_dns: 
2022-02-09 16:10:57,637 DEBUG 1078 log_filename: debug.log
2022-02-09 16:10:57,637 DEBUG 1679 [Device Scan] automatic scan
2022-02-09 16:10:57,637 INFO 1680 Scanning for Shelly devices...
2022-02-09 16:10:58,178 DEBUG 1690 [Device Scan] action queue entry
2022-02-09 16:10:58,228 TRACE 282 RESPONSE: <Response [200]>
2022-02-09 16:10:58,228 TRACE 924 get_security_data
2022-02-09 16:10:58,229 DEBUG 926 [Security] finestasalone.local
2022-02-09 16:10:58,229 DEBUG 935 [Security] username: admin from commandline
2022-02-09 16:10:58,229 DEBUG 936 [Security] password: None from commandline
2022-02-09 16:10:58,230 DEBUG 244 [Device Scan] added: finestasalone.local, IP address: 192.168.1.20, FW Type: homekit, Security Enabled: False to queue
2022-02-09 16:10:58,230 DEBUG 245 
2022-02-09 16:10:58,380 TRACE 370 Device URL: http://192.168.1.112/rpc/Shelly.GetInfoExt
2022-02-09 16:10:58,381 TRACE 371 status code: 401

andyblac, 2022-02-09:
sorry my mistake try:
[flash-shelly.py.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/8033798/flash-shelly.py.zip)

fparodo, 2022-02-09:
2022-02-09 16:18:54,795 TRACE 917 load_security
2022-02-09 16:18:54,796 DEBUG 1050 OS: Darwin
2022-02-09 16:18:54,796 DEBUG 1051 app_version: 3.0.0
2022-02-09 16:18:54,796 DEBUG 1052 do_all: False
2022-02-09 16:18:54,796 DEBUG 1053 dry_run: False
2022-02-09 16:18:54,796 DEBUG 1054 quiet_run: False
2022-02-09 16:18:54,796 DEBUG 1055 silent_run: False
2022-02-09 16:18:54,796 DEBUG 1056 timeout: 20
2022-02-09 16:18:54,796 DEBUG 1057 reboot: False
2022-02-09 16:18:54,797 DEBUG 1058 manual_hosts:  (0)
2022-02-09 16:18:54,797 DEBUG 1059 user: admin
2022-02-09 16:18:54,797 DEBUG 1060 password: None
2022-02-09 16:18:54,797 DEBUG 1061 action: flash
2022-02-09 16:18:54,797 DEBUG 1062 mode: homekit
2022-02-09 16:18:54,797 DEBUG 1063 info_level: 3
2022-02-09 16:18:54,797 DEBUG 1064 fw_type_filter: all
2022-02-09 16:18:54,797 DEBUG 1065 model_type_filter: all
2022-02-09 16:18:54,797 DEBUG 1066 device_name_filter: all
2022-02-09 16:18:54,797 DEBUG 1067 version: 
2022-02-09 16:18:54,797 DEBUG 1068 exclude: 
2022-02-09 16:18:54,797 DEBUG 1069 local_file: 
2022-02-09 16:18:54,797 DEBUG 1070 variant: 
2022-02-09 16:18:54,797 DEBUG 1071 force: False
2022-02-09 16:18:54,797 DEBUG 1072 verbose: 5
2022-02-09 16:18:54,797 DEBUG 1073 hap_setup_code: 
2022-02-09 16:18:54,797 DEBUG 1074 network_type: 
2022-02-09 16:18:54,797 DEBUG 1075 ipv4_ip: 
2022-02-09 16:18:54,797 DEBUG 1076 ipv4_mask: 
2022-02-09 16:18:54,797 DEBUG 1077 ipv4_gw: 
2022-02-09 16:18:54,797 DEBUG 1078 ipv4_dns: 
2022-02-09 16:18:54,798 DEBUG 1079 log_filename: debug.log
2022-02-09 16:18:54,798 DEBUG 1680 [Device Scan] automatic scan
2022-02-09 16:18:54,798 INFO 1681 Scanning for Shelly devices...
2022-02-09 16:18:55,011 DEBUG 245 
2022-02-09 16:18:55,012 DEBUG 227 [Device Scan] found device: finestasalone.local, IP address: 192.168.1.20
2022-02-09 16:18:55,012 TRACE 228 [Device Scan] info: ServiceInfo(type='_http._tcp.local.', name='finestasalone._http._tcp.local.', addresses=[b'\xc0\xa8\x01\x14'], port=80, weight=0, priority=0, server='finestasalone.local.', properties={b'id': b'shellyswitch25-73C352', b'arch': b'esp8266', b'app': b'switch25', b'fw_version': b'2.8.0', b'fw_id': b'20210311-193713/2.8.0-g1e8cd09-master'}, interface_index=None)
2022-02-09 16:18:55,012 TRACE 229 [Device Scan] properties: {'id': 'shellyswitch25-73C352', 'arch': 'esp8266', 'app': 'switch25', 'fw_version': '2.8.0', 'fw_id': '20210311-193713/2.8.0-g1e8cd09-master'}
2022-02-09 16:18:55,017 DEBUG 315 Device: finestasalone.local is Online
2022-02-09 16:18:55,121 TRACE 282 RESPONSE: <Response [404]>
2022-02-09 16:18:55,163 TRACE 282 RESPONSE: <Response [404]>
2022-02-09 16:18:55,218 TRACE 282 RESPONSE: <Response [200]>
2022-02-09 16:18:55,218 TRACE 925 get_security_data
2022-02-09 16:18:55,218 DEBUG 927 [Security] finestasalone.local
2022-02-09 16:18:55,218 DEBUG 936 [Security] username: admin from commandline
2022-02-09 16:18:55,218 DEBUG 937 [Security] password: None from commandline
2022-02-09 16:18:55,218 DEBUG 244 [Device Scan] added: finestasalone.local, IP address: 192.168.1.20, FW Type: homekit, Security Enabled: False to queue
2022-02-09 16:18:55,218 DEBUG 245

andyblac, 2022-02-09:
please post the file, that look incomplete.

fparodo, 2022-02-09:
[debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/8033842/debug.log)

fparodo, 2022-02-09:
./flash-shelly.py finestasalone.local --log-file debug.log
Looking for Shelly devices...

Host: http://finestasalone.local
Device Name: finestasalone
Model: Shelly25
Device ID: shellyswitch25-73C352
Mode: Roller Shutter
SSID: Air
IP: 192.168.1.20
RSSI: -70
Sys Temp: 50˚c
Up Time: 1:25:13
HAP Connections: 0 / 3 / 12
Firmware: HomeKit 2.8.0 ➜ 2.10.1
Do you wish to flash finestasalone to HomeKit firmware version 2.10.1 (y/n) ? y
Downloading Firmware...
Now Flashing HomeKit 2.10.1

flash-shelly version: 3.0.0
Try to update your script, maybe the bug is already fixed!
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/urllib3/response.py", line 438, in _error_catcher
    yield
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/urllib3/response.py", line 519, in read
    data = self._fp.read(amt) if not fp_closed else b""
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/http/client.py", line 458, in read
    n = self.readinto(b)
2022-02-09 16:28:36,229 TRACE 916 load_security
2022-02-09 16:28:36,229 DEBUG 1049 OS: Darwin
2022-02-09 16:28:36,229 DEBUG 1050 app_version: 3.0.0
2022-02-09 16:28:36,229 DEBUG 1051 do_all: False
2022-02-09 16:28:36,229 DEBUG 1052 dry_run: False
2022-02-09 16:28:36,229 DEBUG 1053 quiet_run: False
2022-02-09 16:28:36,229 DEBUG 1054 silent_run: False
2022-02-09 16:28:36,229 DEBUG 1055 timeout: 20
2022-02-09 16:28:36,229 DEBUG 1056 reboot: False
2022-02-09 16:28:36,229 DEBUG 1057 manual_hosts: ['finestasalone.local'] (1)
2022-02-09 16:28:36,229 DEBUG 1058 user: admin
2022-02-09 16:28:36,229 DEBUG 1059 password: None
2022-02-09 16:28:36,230 DEBUG 1060 action: flash
2022-02-09 16:28:36,230 DEBUG 1061 mode: homekit
2022-02-09 16:28:36,230 DEBUG 1062 info_level: 3
2022-02-09 16:28:36,230 DEBUG 1063 fw_type_filter: all
2022-02-09 16:28:36,230 DEBUG 1064 model_type_filter: all
2022-02-09 16:28:36,230 DEBUG 1065 device_name_filter: all
2022-02-09 16:28:36,230 DEBUG 1066 version:
2022-02-09 16:28:36,230 DEBUG 1067 exclude:
2022-02-09 16:28:36,230 DEBUG 1068 local_file:
2022-02-09 16:28:36,230 DEBUG 1069 variant:
2022-02-09 16:28:36,230 DEBUG 1070 force: False
2022-02-09 16:28:36,230 DEBUG 1071 verbose: 5
2022-02-09 16:28:36,230 DEBUG 1072 hap_setup_code:
2022-02-09 16:28:36,230 DEBUG 1073 network_type:
2022-02-09 16:28:36,230 DEBUG 1074 ipv4_ip:
2022-02-09 16:28:36,230 DEBUG 1075 ipv4_mask:
2022-02-09 16:28:36,230 DEBUG 1076 ipv4_gw:
2022-02-09 16:28:36,230 DEBUG 1077 ipv4_dns:
2022-02-09 16:28:36,230 DEBUG 1078 log_filename: debug.log
2022-02-09 16:28:36,230 DEBUG 1661 [Manual Hosts]
2022-02-09 16:28:36,230 INFO 1662 Looking for Shelly devices...
2022-02-09 16:28:36,231 DEBUG 1664
2022-02-09 16:28:36,231 DEBUG 1665 [Manual Hosts] action finestasalone.local
2022-02-09 16:28:36,231 TRACE 924 get_security_data
2022-02-09 16:28:36,231 DEBUG 926 [Security] finestasalone.local
2022-02-09 16:28:36,231 DEBUG 935 [Security] username: admin from commandline
2022-02-09 16:28:36,231 DEBUG 936 [Security] password: None from commandline
2022-02-09 16:28:36,242 DEBUG 315 Device: finestasalone.local is Online
2022-02-09 16:28:36,314 TRACE 282 RESPONSE: <Response [404]>
2022-02-09 16:28:36,394 TRACE 282 RESPONSE: <Response [404]>
[... 184 lines cut ...]
    self.probe_device(device)
  File "./flash-shelly.py", line 1637, in probe_device
    self.parse_info(device, hk_flash_fw_version)  # main convert run, or update stock to latest firmware if an update or color mode is required.
  File "./flash-shelly.py", line 1530, in parse_info
    self.flash_firmware(device)
  File "./flash-shelly.py", line 1294, in flash_firmware
    response = device.write_firmware()  # do actual flash firmware.
  File "./flash-shelly.py", line 662, in write_firmware
    response = requests.post(url=f'http://{self.wifi_ip}/update', auth=HTTPDigestAuth(self.username, self.password), files=files)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests/api.py", line 117, in post
    return request('post', url, data=data, json=json, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests/api.py", line 61, in request
    return session.request(method=method, url=url, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests/sessions.py", line 529, in request
    resp = self.send(prep, **send_kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests/sessions.py", line 687, in send
    r.content
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests/models.py", line 838, in content
    self._content = b''.join(self.iter_content(CONTENT_CHUNK_SIZE)) or b''
  File "/Library/Frameworks/Python.framework/Versions/3.8/lib/python3.8/site-packages/requests/models.py", line 763, in generate
    raise ChunkedEncodingError(e)
requests.exceptions.ChunkedEncodingError: ("Connection broken: ConnectionResetError(54, 'Connection reset by peer')", ConnectionResetError(54, 'Connection reset by peer'))

Devices found: 1 Upgradeable: 1 Flashed: 0
Log file created: debug.log

fparodo, 2022-02-09:
[debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/8033869/debug.log)

andyblac, 2022-02-09:
what does log say on shelly device itself ?

andyblac, 2022-02-09:
what happens if you try to update the device via the device webui?

fparodo, 2022-02-09:
5771986733 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.1.122:51392
5775518983 shelly_main.cpp:554     Up 5775.51, HAP 0/3/12 ns 3, RAM: 18052/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
5783518899 shelly_main.cpp:554     Up 5783.51, HAP 0/3/12 ns 3, RAM: 18240/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
5785264018 mgos_http_server.c:180  0x3fff4a1c HTTP connection from 192.168.1.122:51404
5785308003 mgos_http_server.c:180  0x3fff4a1c HTTP connection from 192.168.1.122:51405
5785324533 mgos_http_server.c:187  0x3fff4a1c GET /settings
5785370978 mgos_http_server.c:180  0x3fff4ad4 HTTP connection from 192.168.1.122:51406
5785384561 mg_rpc.c:285            No handler for Shelly.GetInfoExt
5785421609 mgos_http_server.c:180  0x3fff4ad4 HTTP connection from 192.168.1.122:51407
5785437062 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.1.122:51407
5785548145 mgos_http_server.c:180  0x3fff4ad4 HTTP connection from 192.168.1.122:51408
5785565014 mg_rpc.c:285            No handler for Shelly.GetInfoExt
5785599182 mgos_http_server.c:180  0x3fff741c HTTP connection from 192.168.1.122:51409
5785614248 mg_rpc.c:305            Shelly.GetInfo via HTTP 192.168.1.122:51409
5787750032 mgos_http_server.c:180  0x3fff4ad4 HTTP connection from 192.168.1.122:51411
5787771967 mgos_ota_core.c:252     Starting, timeout 600, commit timeout 0, mem 13588
5787806119 mgos_ota_core.c:486     FW: switch25 esp8266 2.10.1 20211016-123111/2.10.1-gc8923b3-master
5787815956 shelly_main.cpp:365     == Stopping HAP service (1)
5787826730 mgos_ota_core.c:587     begin: waiting for user
5787838728 shelly_main.cpp:376     HAP server state: 2
5787854745 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff718c nc 0x3fff75d4 0/2/12
5787904486 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6c7c nc 0x3fff6d2c 0/1/12
5787918224 HAPPlatformTCPStrea:303 HAPPlatformTCPStreamClose ts 0x3fff6e3c nc 0x3fff6db4 0/0/12
5787933546 shelly_main.cpp:365     == Stopping HAP service (2)
5787942615 mgos_ota_core.c:587     begin: waiting for user
5787961166 HAPPlatformTCPStrea:105 0x3fff7004 0.0.0.0:9012 HAP connection closed, ns 0/0/12 i 1
5787974262 shelly_main.cpp:376     HAP server state: 0
5788008607 mgos_sys_config.c:232   Loading conf2.json
5788161772 mgos_sys_config.c:174   Saved to conf9.json
5788210497 mgos_mongoose.c:66      New heap free LWM: 10728
5788224674 shelly_main.cpp:781     Starting firmware update
5788241782 esp_ota_backend.c:185   Slot 1, FW: switch25.bin -> 0x108000, FS fs.bin -> 0x1bb000
5788260397 mgos_ota_core.c:504     0.18% total, esp_init_data_default_v08.bin 128 of 128
5788399524 esp_ota_backend.c:338   Start writing fs.bin (262144) @ 0x1bb000
5788446858 mgos_ota_core.c:504     0.23% total, fs.bin 512 of 262144
5791514731 shelly_main.cpp:554     Up 5791.51, HAP 0/0/12 ns 0, RAM: 17060/38320; st 50; disabled
5792354899 esp_ota_backend.c:224   SHA1 262144 @ 0x1bb000 = 7bd21b89431ce53e853a5965800486a3fae0cb1e, want 5c045317ac8f0d7ae95bee1db065686cd9be367b
5792367440 mgos_ota_core.c:811     Update finished, result -2 (Invalid checksum)
5792508782 shelly_main.cpp:311     === Creating accessories
5792558714 shelly_hap_window_c:169 WC 1: mp 131.84, mt_ms 28490, cur_pos 76.99
5792569970 shelly_main.cpp:341     === Starting HAP server (CN 1)
5792639740 HAPPlatformTCPStrea:187 Listening on 9013
5792655224 shelly_main.cpp:376     HAP server state: 1
5792767487 HAPPlatformTCPStrea:121 0x3fff70d4 192.168.1.195:50934 Incoming HAP connection, ns 0/0/12
5792783283 HAPPlatformTCPStrea:276 0x3fff70d4 192.168.1.195:50934 Accepted HAP connection, ns 0/1/12 ts 0x3fff5da4
5794273587 HAPPlatformTCPStrea:121 0x3fff78bc 192.168.1.197:49879 Incoming HAP connection, ns 0/1/12
5795825946 HAPPlatformTCPStrea:276 0x3fff78bc 192.168.1.197:49879 Accepted HAP connection, ns 0/2/12 ts 0x3fff6e14
5799518984 shelly_main.cpp:554     Up 5799.51, HAP 0/2/12 ns 2, RAM: 18436/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0
5804853017 HAPPlatformTCPStrea:121 0x3fff648c 192.168.1.214:53903 Incoming HAP connection, ns 0/2/12
5804868657 HAPPlatformTCPStrea:276 0x3fff648c 192.168.1.214:53903 Accepted HAP connection, ns 0/3/12 ts 0x3fff718c
5807519108 shelly_main.cpp:554     Up 5807.51, HAP 0/3/12 ns 3, RAM: 17764/38320; st 50; 4.1: c:1 mp:131.84 mt_ms:28490 cp:76.99 tp:76.99 lemd:0 lhmd:0

fparodo, 2022-02-09:
and from UI "Error uploading: TypeError: Failed to fetch"

andyblac, 2022-02-09:
ok it seem to be a device issue, @rojer can you help ?

```
5788224674 shelly_main.cpp:781 Starting firmware update
5788241782 esp_ota_backend.c:185 Slot 1, FW: switch25.bin -> 0x108000, FS fs.bin -> 0x1bb000
5788260397 mgos_ota_core.c:504 0.18% total, esp_init_data_default_v08.bin 128 of 128
5788399524 esp_ota_backend.c:338 Start writing fs.bin (262144) @ 0x1bb000
5788446858 mgos_ota_core.c:504 0.23% total, fs.bin 512 of 262144
5791514731 shelly_main.cpp:554 Up 5791.51, HAP 0/0/12 ns 0, RAM: 17060/38320; st 50; disabled
5792354899 esp_ota_backend.c:224 SHA1 262144 @ 0x1bb000 = 7bd21b89431ce53e853a5965800486a3fae0cb1e, want 5c045317ac8f0d7ae95bee1db065686cd9be367b
5792367440 mgos_ota_core.c:811 Update finished, result -2 (Invalid checksum)
```

timoschilling, 2022-03-17:
@fparodo can you please try to update via web ui from your 2.8.0 to 2.8.1 and then to 2.8.2 and then to 2.9.0. You can find the versions here https://rojer.me/files/shelly/

closed 2022-04-25 by github-actions[bot]