mongoose-os-apps/shelly-homekit #930: Shelly 2.5 not showing up in wi-fi

davidpraznik, 2022-02-11:
I mean .. everything worked ok. I tried different roller shutter settings where shelly 2.5 always requires calibration. All of a sudden the roller shutter has stopped working (probably the shutter's temperature fuse since it works OK now). I turned off and back on the main power fuse but there was no effect. Then I noticed that I no longer have access to the web interface on http://192.168.0.*/ where I edited the settings. Then I did a hard reset on a shelly but I still don't see it in my wi-fi network and I can't add or configure it. 
Shelly 2.5 is flashed with the latest shelly-homekit firmware.  

How can I solve this problem? 
Thank you very much, 
David

timoschilling, 2022-02-11:
join the `shellyswitch25-***` wifi and open 192.168.33.1 in your browser, then you can reconfigure the shelly.

davidpraznik, 2022-02-11:
@timoschilling oh boy, how very grateful I am to you. Danke sehr!! 😄

closed 2022-02-11 by rojer