mongoose-os-apps/shelly-homekit #934: Keep single switch on/off toggle and expose double an long-click to Home as button

vifi42, 2022-02-14:
Hey :) 
This might be a niche request, but in the event of my network going down or my AppleTV/Homepod losing power I'd love to be able to still use my switches and lights.
BUT I'd also love to use the singe/double/long press function in Home for the switches.
So is there a way to create a "combined" switch setting (for momentary switches of course) that combines the on-off-toggle for single press but presents the double and long press to HomeKit (maybe even als separate buttons?)
This would be awesome!
Thanks a lot!

andyblac, 2022-02-19:
double of https://github.com/mongoose-os-apps/shelly-homekit/issues/829

timoschilling, 2022-02-19:
@andyblac i think you referenced the wrong issue/pr #147 hasn’t merged anything. 

I also think this request isn’t covered by #9. it requested a mix of switch and stateless input, which we don’t have yet

timoschilling, 2022-02-19:
But anyway this is a double of #829.

closed 2022-02-19 by timoschilling

andyblac, 2022-02-19:
> @andyblac i think you referenced the wrong issue/pr #147 hasn’t merged anything.
> 
> I also think this request isn’t covered by #9. it requested a mix of switch and stateless input, which we don’t have yet

see https://github.com/mongoose-os-apps/shelly-homekit/issues/320

timoschilling, 2022-02-19:
In #320 it was requested as a bug fix, here it is requested as a feature.