mongoose-os-apps/shelly-homekit #936: mac address

Warner1970, 2022-02-14:
I've been searching but I can't seem to find this: Is there any way I can read the mac address of the Shelly 1? I would like to make reservations in my DHCP server on mac address.
I could use client ID but I seems my DHCP is a bit buggy when using client ID

timoschilling, 2022-02-14:
We don't display them at the moment, but there is a hack: Change the name of the shelly and append `-????????????` (12 x ?) and save it, maybe you need reboot the device. The question marks will be replaced with the Mac address of the shelly. Then you simply need to place a : after each second char.

Warner1970, 2022-02-15:
Thanks Timo, I'll give it a try. But I also see an enhancement coming up. Great

closed 2022-02-15 by Warner1970

timoschilling, 2022-02-15:
There is already a way to read the MAC address without hack! In the Home app, in the accessory informations is the serial number displayed which is equal to the MAC address.

Warner1970, 2022-02-16:
Yes, found it.. perfect. thanks!

PR #937 (2022-02-18): MAC in UI
closes #936 
[image]