mongoose-os-apps/shelly-homekit #939: POC: Bulb reset

timoschilling, 2022-02-18:
@rojer my idea for a bulb reset mechanism is to add a `shelly.reboot_counter = 0` to the bulbs config.
```cpp
void InitApp() {
  if (mgos_sys_config-> shelly_reboot_counter >= 3) {
    DoReset();
  }
  mgos_sys_config-> shelly_reboot_counter++;
  mgos_sys_config_save(&mgos_sys_config, false, nullptr)
  mgos_set_timer(10000, 0, ResetRebootCounter, NULL);
```

What do you think about it?
Should we only add it to bulbs or to all shelly's, I see a benefit having it on all shelly's, sometime use have shelly's without switches connected behind a wall and can't reset them.

MilanBacaCz, 2022-02-18:
It is greate idea. I have many shellys. They work well, but sometimes they are unresponsive and i have to Power off them. Then they work well again. I tried reconnect them in the router, but it did not help. Only Power off and on make them work again. So reset timer is good idea.

MagOO33, 2022-02-19:
Great!
same kind of idea, could it be possible to store the last command in order to show the good state if unwanted reboot appears because of power spikes.
The case is shelly 2.5 controlling roller shutters and launching a up or down command and rebooting because of inductive spike.
When it reboots, it shows the wrong state because command was done by relay while rebooting.
This could avoid a calibration to make roller shutter rework

timoschilling, 2022-02-23:
closed because PR is open

closed 2022-02-23 by timoschilling