mongoose-os-apps/shelly-homekit #943: Automation stopped working

FlorianTDK, 2022-02-25:
Hello,

I've got an issue on my Shelly 2.5. I've got 2 of them, both for roller shutter. I've got them from almost a year now and everything was working perfectly. Since day 1 I've got automation (one in the morning to open them, one in the evening to close them).
Few weeks ago, my automation suddenly stopped working. It works when I open/close them from the HomeKit app or from the WebUI, but any automation works.
I've got an AppleTV as hub so I've followed the step : deactivate it, re-create the automation and so on. Nothing works. I've also totally reset one the Shelly, remove it from HomeKit, re-paired it. Same issue.

I'm still on the last stable version (2.10.1 - build: 20211016-123111/2.10.1-gc8923b3-master) on one of them and I just tried today to install the latest beta (2.11.0-beta1 - build:20220208-023817/2.10.1-60-g0060726-master) to check if it was fixing something. But nothing changed. 

Nothing changed in my network, I've got static IP on my router for all my connected device since years, everything correctly set up.
So I really don't know.

Does someone face the same issue? 
I can run some test to help find the issue. Thanks a lot.

This is the number 1 feature for me, so I hope someone will help me.

timoschilling, 2022-02-25:
If you are able to control your Shelly from the iPhone, the issue must be on the HomeKit site. In HomeKit Accessoires like the Shelly didn’t have any knowledge about automatization, they just got control command’s from Apple devices. Maybe you have a connection issue between your Apple TV and the Shelly? Can you control your Shelly when you are on cellular and not on WiFi?

FlorianTDK, 2022-02-26:
Ok, ok sooo @timoschilling thanks a lot for your comment.
Because I didn't try in fact on cellular. Since your comment I was struggling on WHY it was not working in cellular if I've got an hub!!!
My Apple TV was connected in Ethernet to my main router and all the other device to WiFi to my second router. It was working fine until few days (why it stopped? idk) but switching the ethernet port for master router to slave router made it worked perfectly !
So, it was a network issue.. Which I can not explain, but your comment was helpful still :)

closed 2022-02-26 by FlorianTDK