mongoose-os-apps/shelly-homekit #944: Having issues when updating my Asus router

jmartinezs77, 2022-02-26:
Running my wifi on Asus router xt8 I have plenty of problems with shelly 2.5 after router firmware update. web server unavailable for several minutes, HomeKit accessories not responding for minutes. Seems to be a blockage somewhere.
Same issue transmitted to Asus.
But just wanted to let Rojer know as long as could be improved through wifi libraries used or whatever.
I am considering changing shelly firmware to original as long as this issue is desperation me.

timoschilling, 2022-02-26:
Stock and this firmware are using the same WiFi lib. You can try the 2.11 beta (see the test 2.11 test issue for details) or you downgrade to 2.9 to test with different WiFi lib versions

rojer, 2022-02-27:
there haven't been significant changes to the wireless stack in recent versions, only some minor stability fixes.
entire SDK from espressif that we use is in maintenance mode and they are very reluctant to make any changes, unfortunately...
that said, if there's complete loss of connectivity, it's something they may address.

timoschilling, 2022-03-10:
Looks like this is really a Asus router bug, which Asus is already working on: https://www.facebook.com/groups/ShellyIoTCommunitySupport/permalink/4711940455571948/

I will close this as not related to this firmware.

closed 2022-03-10 by timoschilling