mongoose-os-apps/shelly-homekit #945: Shelly "Lock" as a security related device

Multisaft7, 2022-02-27:
Hey together,

I am using a Shelly 1 Plus as a switch for my automatic dooropening function of my new front door.
As now I can open it via Siri but this is only possible cause the homekit does not recognize the shelly as a verified security device.

Is there a way to change this or make the shelly homekit firmware "dummy" certificated? That any people at my home are able to just open the door via voice?!

Hope you understand my problem!?

Cheers

timoschilling, 2022-02-27:
I think you are mixing up different things, but let’s try to figure out what your problem is. 

1. please post a screenshot from you Shelly switch settings
2. have you used an iPhone or Apple Watch to unlock? If it was an iPhone was it locked or unlocked?
3. Same question as 2. but for the other person in your household
4. Was the other person connected to your WiFi as he has tried to open it?

> Is there a way to change this or make the shelly homekit firmware "dummy" certificated?

no, that’s not possible, but that is definitely not your problem! The only difference between certified and non certified devices is the warning and that certified devices can link to there app.

Multisaft7, 2022-02-27:
1.
[image]
2. I used the iPhone or Apple Watch via Siri to unlock the door!
3. Yes my wife should also use this
4. I am able to unlock the door when i am not in the wifi...

But I think I mean the warning... That no one of my friends is able to open the doors via voice over my homepods.
Also I do not want to just be able to open the door directly via voice...

timoschilling, 2022-02-27:
Did you mean this message?
[image]

It only means that we are not a official licensed product and that the communication between HomeKit and the Shelly is maybe insecure. It doesn't effect the functionality.

About the questions:
1. Ok
2. The main point of the question is "If it was an iPhone was it locked or unlocked?". Because only unlocked devices can open secured devices.
3. Same as 2.
4. The question was are other people able to control devices when not on your wifi. Thats a config option in HomeKit.

A lock is a secured devices in HomeKit, a window cover for example not. You can only open secured devices on authenticated devices. For example an Apple Watch is alway an authenticated device after you put in the code until you removed it from your wrist. A iPhone is only a authenticated devices if it is unlocked. HomePods are unauthenticated devices and can not open secured devices.

tl;dr: If you are able to control the lock via your iPhone, but not on the iPhone from some else or on a HomePod, the problem is out of scope for us. The devices didn't have any knowledge about iPhone, HomePod or Users, they only get control commands or publishing a state.

closed 2022-02-27 by timoschilling