mongoose-os-apps/shelly-homekit #947: Automations fails after first time

FranciRoosters, 2022-03-02:
I have an automation that should turn on my closet light when the door opens and one that should turn the closet light off when the door closes. The automations, as of late, has been very unreliable and it just works the first time I open and close the door. After that, it fails every time. I’m on the latest firmware, have restarted my hub, deleted the automations and recreated them, but nothing. What else can I try? I wouldn’t really want to kill of the power for this. Thank you.

timoschilling, 2022-03-02:
Is your device responsible when you control it via iPhone?

FranciRoosters, 2022-03-02:
It is. I also reset the Eve door sensor, but nothing.

timoschilling, 2022-03-02:
Then it isn’t a problem with the Shelly, it didn’t have any knowledge about automations, it only get commands. Maybe you have a connection issue between your Shelly and your hap

closed 2022-03-02 by timoschilling

FranciRoosters, 2022-03-02:
What’s a hap?

timoschilling, 2022-03-02:
Sorry typo, I mean hub