mongoose-os-apps/shelly-homekit #950: Question: how to keep multiple Shellys' statuses in sync (e.g. via Direct Device to Device Communication)

seidnerj, 2022-03-09:
Does the firmware currently support (and how?) keeping multiple Shellys' statuses in sync, e.g. ON/OFF?

My scenario is like so:

1. I have two Shelly 1s, one at the beginning of a hallway and the other one at the end of the hallway. They are both connected to stateless "momentary" switches with a small indicator LED light. Let's call these A and B.

2. I have another Shelly 1, connected to a light bulb. Let's call this one X.

3. When I click either A or B, depending on their state, they either turn X on or off - keeping it in sync. In addition, if A is on, it makes sure B is on and vice versa. Bottom line, A, B and X are all kept in sync.

Currently, I've implemented this with the native Shelly firmware by triggering the relevant URLs via Device to Device Communication but it seems this is not implemented in the Shelly-Homekit firmware.

The reason I need to do this is that A, B and X are not on the same circuit so I cannot implement this in the usual way. If I had the proper wiring I believe I could do this with a single Shelly.

Is there a wait to implement something this with the Shelly-Homekit firmware? I hate having the connect Shelly via the Homebridge plugin and rely on the Shelly cloud.

Thanks in advance!

andyblac, 2022-03-09:
You could do this via HomeKit automations.

andyblac, 2022-03-09:
the easiest was would be to use the shelly1 A and  B in "Detached" mode, then create a toggle automation to turn on / off X

see here https://github.com/mongoose-os-apps/shelly-homekit/wiki/Stateless-Toggle-Switch

seidnerj, 2022-03-09:
I do need the state in A and B because they are both connected to a switch that has an LED indicator light and I want that LED light to be on when X is on. So I guess what I could do is set both A and B to momentary switches and connect their output to the LED light and then sync A, B and Xs states via automations. The only downside would be reliance on a home hub.

Did I get this right?

andyblac, 2022-03-09:
just add the LED to shortcut action, so it turns on / off as well. so you would have 2 IF blocks, one for X and one for LED,

Yes you would need a hub, atm this is your only solution, as inter-device connectivity has not been coded yet, it is no the todo list, but we have no time frame for it atm.

seidnerj, 2022-03-09:
The idea is that the A, B an X are all either ON or OFF while A and Bs respective LED lights are in sync as well.

I.e., when the light is on (via X), then A and B (and their respective lights) are all on. When the light is off, they are all off. The reason this is important is that I have another scenario where the switch is far from the actual light and I want to be able to look at the switch and know whether or not the light is actually on.

A shortcut would be dependent on an iPhone/iPad or similar being present, no? If not, then I'm not aware of a way to define a shortcut at the hub level - how is this done?

At any rate, I think what you suggested combined with what I wrote before, i.e. connecting to the indicator LED light to the output of the Shellys behind the switches and then syncing all 3 via automations gives me a good way forward. 

Many thanks!

andyblac, 2022-03-09:
yes, you would need a "Hub", old iPad, AppleTV or HomePod, HomePod mini, is cheapest option £89 in uk, can be found at times for £69

timoschilling, 2022-03-09:
I think we can close this, because it's more a question about how to do automatisations then over how a shelly works.

closed 2022-03-09 by timoschilling