mongoose-os-apps/shelly-homekit #953: Shelly 2.5 roller shutter calibration fail

AzRowD, 2022-03-10:
Hi,
My Shelly 2.5 work perfectly with the stock firmware but when I put the homekit firmware the calibration doesn't work properly. The roller shutter go down during about 0.5 second then go up during 0.5sec and stop... So when I Close or I Open my roller shutter, it move about 1centimeter haha. I don't know if it's an issue or a bad configuration. Can you help me ? Please see my settings in attachements.
Best Regards?
AZD
[image]
[image]

timoschilling, 2022-03-10:
can you please test it with our 2.11 beta, download this zip file, don't open it, just upload it via UI
https://rojer.me/files/shelly/2.11.0-beta3/shelly-homekit-Shelly25.zip

AzRowD, 2022-03-11:
Hi @timoschilling,
Thank you for this fast reply, this firmware fix my issue. Calibration work perfectly now, so nice !  Good Job Guys 👍 
So I have also one request/question about my physical switch . Currently I have 3 btn UP/STOP/DOWN. With the Shelly stock firmware the STOP btn work so nice with the config below (in the UI and physicaly), but with the homekit firmware 2.11.O beta it doesn't work and does nothing. Also In the UI I can't see any STOP btn. Is it normal ? or I need some configuration ?

[image]
[image]

Thank you,
Best Regards

AZD

timoschilling, 2022-03-11:
Nice that 2.11 works for you.

About the STOP button, we didn't support it at the moment, there is already a feature request open for that. Because of that I will close this issue.

closed 2022-03-11 by timoschilling