mongoose-os-apps/shelly-homekit #954: “Garage Door Obstructed” error status message

coreygreenberg, 2022-03-11:
Shelly 1 flashed to Shelly-Homekit
Reed switch connected as Normally Open

This setup works properly as a Homekit garage door opener, but when the door is closed, the Shelly 1 gives a status message error “Garage Door Obstructed”. As the system opens and closes the garage door perfectly this bug is certainly livable, but is there a setting I am blind-spotting that would replace this status error message with something more appropriate like “Door Closed”?

Frankly I’m amazed I could Siri-enable my 20 yr-old garage door with a $15 switch. But am I missing something in the settings, whether in Shelly-Homekit or Homekit itself? Thank you for your help.

coreygreenberg, 2022-03-14:
@MagOO33 Thank you! That sorted everything. Now not only is the status displayed properly, without the obstruction error message, but for some reason my garage door commands are showing up in CarPlay dashboard for the first time, as if they had refused to show up there until I got the status switch set up properly. Thank you thank you thank you Mr. Magoo!

MagOO33, 2022-03-14:
you're welcome. Glad to here

closed 2022-03-14 by timoschilling