mongoose-os-apps/shelly-homekit #956: roller shutter shelly 2.5 with 3 buttons up, down and stop in homekit?

marco57330, 2022-03-12:
hello team,
i did not find my issue in the list, maybe i did something wrong but already spent several hours on it.
my issue is simple: i would like to add in homekit the roller shutter managed by 3 buttons: up, down and stop which is the actual switch i have, when i am adding it to your interface, there is only 2 buttons up and down but not the "middle" stop button. so when importing to homekit the 3 buttons are not there even if i well defined them in the shely app.
if you could help it will be great :)))
thank you in advance

MagOO33, 2022-03-15:
Hello, it works perfectly for me in "separate - toggle" mode.
I have somfy 3 buttons  (mechanical)

marco57330, 2022-03-15:
hello,ok thank you for your answer. i tried several options in your interface but only 2 buttons are displayed (up and down) but not the "stop" button. i did not take screenshot as i reverse to the initial system after having spent 2 hours trying to play and import in homekit.is there a possibility to import in homekit from shelly cloud app without crashing/deleting the shelly app setting performed for all my devices?thanks
Marc-Olivier ***@***.***

MagOO33, 2022-03-15:
Homekit never shows 3 buttons, only a slide (touch and stay to slide)

marco57330, 2022-03-15:
ok thank you for your fast answer. Is there a possibility to import in homekit from shelly cloud app without crashing/deleting the shelly app setting performed for all my devices? (in case i am doing a mistake i do not want to crash the work performed on the shelly app)thank you in advance
Marc-Olivier ***@***.***

MagOO33, 2022-03-15:
I never used the shelly app.
I don't think it's possible because HomeKit have really basic settings, different from shelly ones.

marco57330, 2022-03-15:
ok thank you for your answer. i will see if i go to homekit as when i did it crashed all my shelly data in the app and i do not want to spend more time on it.thank you for your support and nice end of day,
Marc-Olivier ***@***.***