mongoose-os-apps/shelly-homekit #958: Unable to convert Shelly 2.5

MrSimonWood, 2022-03-14:
Hello, I've successfully updated a few Shelly 2.5s before. Installed a new one, updated stock firmware to 20220209-093016/v1.11.8-g8c7bb8d  and tried to update using the link as I have before
http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly25.zip

I get the message:
{"status":"updating","has_update":false,"new_version":"20220209-093016/v1.11.8-g8c7bb8d","old_version":"20220209-093016/v1.11.8-g8c7bb8d"}

And when I go back to the device's IP address, it's still the stock firmware, not the HomeKit version. This is from the log:

`1233369661 shelly_update.c:168     Starting ota from http://rojer.me/files/shelly/shelly-homekit-Shelly25.zip
1233877091 mgos_ota_http_clien:270 Update URL: http://rojer.me/files/shelly/shelly-homekit-Shelly25.zip
1233996482 mgos_ota_http_clien:144 Got redirect: [https://rojer.me/files/shelly/shelly-homekit-Shelly25.zip]
1234002788 mgos_ota_http_clien:270 Update URL: https://rojer.me/files/shelly/shelly-homekit-Shelly25.zip
1234020375 mgos_ota_http_clien:102 Failed to connect
1234076431 shelly_update.c:111     SHOTA: error Failed to connect
1234082436 shelly_update.c:130     OTA is fubar`

The device appears to be able to connect to the internet as it updated the firmware, so I'm not sure why it's failing to connect for the HomeKit firmware. I'm not sure what else to do to troubleshoot... I'd be grateful for any suggestions, thank you!

timoschilling, 2022-03-14:
Can you please retry it, there was a other user reporting the same, his problem has solved by retry.

closed 2022-03-14 by rojer

ruebyi, 2025-01-01:
Happy new year!
I'm running in exact the same problem right now - anything I can do from my side?