mongoose-os-apps/shelly-homekit #961: shelly 25 cannot convert

Jannes80, 2022-03-16:
hello, I have the following problem:
I can't convert the shelly25 via the link: [
](http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly25.zip
I get this error message:
{"status":"updating","has_update":false,"new_version":"","old_version":"20220209-093016/v1.11.8-g8c7bb8d"}

This is from the log:
3342795827 shelly_update.c:168     Starting ota from http://rojer.me/files/shelly/shelly-homekit-Shelly25.zip
3343301812 mgos_ota_http_clien:270 Update URL: http://rojer.me/files/shelly/shelly-homekit-Shelly25.zip
3345687486 mgos_http_server.c:180  0x3fff27d4 HTTP connection from 192.168.1.215:52680
3345712488 json.c:426              RAM: 49920 total, 32780 free
3350706984 mgos_http_server.c:180  0x3fff56fc HTTP connection from 192.168.1.215:52681
3350733063 json.c:426              RAM: 49920 total, 32600 free
3352037686 shelly_sntp.c:433       minute tick at 16:25:00
3353031261 mongoose.c:12126        Failed to resolve 'rojer.me', server 8.8.8.8
3353037408 mgos_ota_http_clien:123 Connect error: -1
3353052169 mgos_ota_http_clien:102 Connect error
3353057051 shelly_update.c:111     SHOTA: error Connect error
3353061843 shelly_update.c:130     OTA is fubar

thanks for support

Byronsasvin, 2022-03-16:
mismo problema y estoy intentando flashearlo con python y dice que espera el reinicio y no hace nada

PS C:\Users\ETS\Downloads\shelly-homekit-master\shelly-homekit-master\tools> ./flash-shelly.py -a
Scanning for Shelly devices...

Host: http://shellyswitch25-E8DB84D2EE75.local
Device Name: Cocina
Model: Shelly25
Firmware: Stock 1.11.8 ➜ HomeKit 2.10.1
Do you wish to flash shellyswitch25-E8DB84D2EE75 to HomeKit firmware version 2.10.1 (y/n) ? y
Now Flashing HomeKit 2.10.1
**waiting for shellyswitch25-E8DB84D2EE75 to reboot.**
Failed to flash shellyswitch25-E8DB84D2EE75 to HomeKit 2.10.1

Devices found: 1 Upgradeable: 1 Flashed: 0 Failed: 1
PS C:\Users\ETS\Downloads\shelly-homekit-master\shelly-homekit-master\tools>

timoschilling, 2022-03-16:
> 3353031261 mongoose.c:12126 Failed to resolve 'rojer.me', server 8.8.8.8

Looks like the google DNS had some issues to resolve the domain name rojer.me, please try it again.

timoschilling, 2022-03-17:
Please answer if it's still a issue, I will reopen it then.

closed 2022-03-17 by timoschilling