mongoose-os-apps/shelly-homekit #962: Since tvOS 15.4 - only Shutters with Shelly 2.5 are not reachable

Rappelsack, 2022-03-17:
Hello,

since tvOS 15.4 / iOS 15.4 the shutters with Shelly 2.5 are not reachable any more.

Just in Home. I can connect and control them via the webpage via IP address. And it shows connections in the homekit section.
Nothing changed except the update to tvOS 15.4 of my homepods.

Now there are month since the last update to the actual 2.10.1 shelly FW. Is there no more development?

Kind rgds

timoschilling, 2022-03-17:
I can confirm that tvOS 15.4 brings problems. I was able to fixe my issue by disabling the HomeHub function on my Apple TVs, so that my HomePods take over. 

That showed my that it is a bug on Apples side. Because of that I will close this issue. 

2.11 will come in the next weeks

andyblac, 2022-03-17:
IMO tvOS should not be used as Home Hub, it brings nothing but issues. It should only be used if no other option.

Rappelsack, 2022-03-17:
You know that tvOS is on the HomePods?

I do not have a Apple TV.. Just 6 HomePods.

Rappelsack, 2022-03-17:
And this issue comes along with tvOS 15.4 on a Homepod only Homekit system.

andyblac, 2022-03-17:
@Rappelsack no longer true it now has it own OS, due to HomePod mini, and Thread support. I have previously updated my HomePod software before tvOS was available.

Rappelsack, 2022-03-17:
Okay, but it doesn't matter because the problem is given and it is on a Homepod only system.

andyblac, 2022-03-17:
if your having issues, try turning off all HomePods, and just try the main one you want to act as master hub, I have had this before when updating OS's, once master is online wait 5 mins, then try again, if works ok, power up all other HomePod's

Rappelsack, 2022-03-17:
Did this already and I am not the only one who has this problem. There are other users with the shelly HK FW who has this problem in a german apple forum.

andyblac, 2022-03-17:
tbh, if it just started with 15.4, it sounds like a OS issue not a FW here as nothing has changed here.

andyblac, 2022-03-17:
i.e HK software is still broken for Stateless buttons with Security System, you can't use IF armed do this else do that. used to work in 12.x been broken since 13.x still broken. Apple breaks shit all the time and does not give a shit to fix it.

Rappelsack, 2022-03-17:
Maybe a code changed in 15.4 an needs to be updated with 2.11?
Neither automations nor "hey Siri" will work. And in home they are "updating status" all the time.
The webinterface shows me under Homekit: 0 pending, 1 active, 12 max.
And they show the correct position "idle pos 100" and they work if I open or close them via the web interface.

I will try to switch off 5 Homepods now.

nikwest, 2022-03-17:
I'm seeing the same issues both on my two homes one with HomePod only and one with HomePod and AppleTV. Both show all devices not reachable, but actually I can access them through their http interface without any issues.
Rebooting the HomePod solves the issue, but after a while the problem ist back.
Maybe 15.4 introduced an incompatibility with our implementation? I somewhat doubt that this is a general problem with 15.4. Otherwise Apple QA should have seen this before releasing it?

timoschilling, 2022-03-17:
Sorry but this is not a Shelly HomeKit issue, it's a xOS 15.4 issue. Let me explain some things:
1. > Maybe a code changed in 15.4 an needs to be updated with 2.11?

   or
   > Maybe 15.4 introduced an incompatibility with our implementation?

   We use Apples own base implementation and didn't have a own implementation of HAP. And that implementations is up to date, there is no new commit since 1/2 year. That HAP protocol, which is used for the communication between xOS and the accessory is build to not be changed, because it should not break devices.
   
2. > The webinterface shows me under Homekit: 0 pending, 1 active, 12 max.

    Yes I can confirm that, but for me it looks like a change in 15.4 that only the HomeHub keeps an open connection to the accessories (I can see it in the debug info's). There is another change which I can observe, which confirms my assumption. Since 15.4 I can turn on a light from my iPhone without connecting to the bulb. Before 15.4 the iPhone has tried to connect to the bulb and them turned them on, now it instantly turns on the bulb. That change totally makes sense because it saves battery power on mobile devices. I expect that they made a change in HK at all, that primary only the hub keeps connected. 
    
3.  > I somewhat doubt that this is a general problem with 15.4. Otherwise Apple QA should have seen this before releasing it?

     That would be the first time that Apple introduces a bug in HK
     
4. > All worked and works fine, now except the shutters with the shelly 2.5 since 15.4.

    Thats a common issue even with official shutters as well.
    https://smartapfel.de/probleme-an-allen-fronten-siri-verweigert-den-dienst-spam-und-kaputte-schalter/
    https://www.reddit.com/r/HomeKit/comments/te7s4u/ios_154_siri_no_longer_understands_how_to_operate/

I know it's frustrating that your smart home isn't working anymore, but it's not our fault!
We will keep this issue close until someone brings an evidence that it is really a issue on our side!

closed 2022-03-17 by timoschilling