mongoose-os-apps/shelly-homekit #966: Input bounce on two way switching - RGBW2

digbycampbell, 2022-03-20:
Hello,

First off thanks for excellent firmware! I am using the 2.11.0 beta though I saw the below behavior on stable release.

I wanted to raise an issue I am having with my RGBW2. I am using two way switching connected to the input and the input is bouncing when I have the switches in 10 or 01 state. Usually, I would consider this to be a hardware issue with my wiring configuration but this behavior is not present when I use the stock firmware.

I imagine there is some leakage voltage through the two way switching - and I suppose that the stock firmware has some debounce or minimum pickup on the input. I thought this might be a potential quality improvement (and would allow me to use my two way switching :) )

I have included a log [log.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/8310755/log.txt) or go here: https://pastebin.com/HmJLhiyA

Thanks again!

timoschilling, 2022-04-20:
Are you using main voltage on the switch and this AC/DC adapter which is included in the rgbw2 package?
If yes, you need to change your whirring, we don't support it at the moment.

digbycampbell, 2022-04-20:
Ah yes. You're right - I am using the AC/DC adapter due to the available wiring in my house.

I'll take another look and see if there's a way around it.

Thanks for the response 😃

closed 2022-04-21 by timoschilling