mongoose-os-apps/shelly-homekit #968: Homekit label layout

lalolerry, 2022-03-20:
Hallo everybody from Italy,
I'm a newbie and not a programmer, but I really appreciate your works and this shelly-homekit firmware.
Everything is easy and work flowless, but I've a very little enanchment to suggest.
It is possible to change the QR homekit code layout to look like the orignal Apple one, as used in its HomeKit Accessory Simulator app?
I mean showing out in this way:
[image]

thanks!

rojer, 2022-03-20:
why?

rojer, 2022-03-20:
> Because I use to print it on a label and stich it to the devices (like natively supported homekit devices).

sorry, missed that part. well, i guess it's possible. if you can contribute the SVG code to draw the outline and the little house, we can try to incorporate it. not sure what to do about the numbers yet.

lalolerry, 2022-03-20:
Wow, what a quick reply! I'm really honored to speak with you, Rojer!
Unfortunatelly I'm not good in programming at all, I can only donate for this lovely project (as I already did). :/

lalolerry, 2022-03-20:
> sorry, missed that part. well, i guess it's possible. if you can contribute the SVG code to draw the outline and the little house, we can try to incorporate it. not sure what to do about the numbers [...]

ehm actually I edit later adding this part ;P

alexbartsch, 2022-03-31:
Hey @rojer and @lalolerry. I'm pretty new to this smart home thing, but I'm a developer for many years. Especially in web development. Let me see what I can do. 😉

PR #973 (2022-04-02): HomeKit label layout
resolves #968

Here is how it looks like:
[image]
closed 2022-04-02 by timoschilling