mongoose-os-apps/shelly-homekit #977: Can't Flash Shelly Dimmer 2 with HomeKit

Imburdalinix, 2022-04-23:
So today I tried to flash my shelly dimmer 2 with HomeKit. When trying so the link didn't do its job. Here the link I used to flash the shelly:

http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly2.zip

I replaced the A.B.C.D with the IP of the shelly. It did something because it outputted this:

{"status":"updating","has_update":false,"new_version":"20211109-131033/v1.11.7-g682a0db","old_version":"20211109-131033/v1.11.7-g682a0db","beta_version":"20220228-121849/v1.11.9_Dimmer_RC1-g84f0b8b"}

I waited about 5 to 10 minutes and when I entered the IP of the shelly into my webbrowser it just showed me the normal Shelly interface and not the HomeKit interface...

I am really confused and don't know why this didn't work. If someone could help me it would be very nice.

Just in case somebody needs my current Firmware Version of my shelly here it is:

20211109-131033/v1.11.7-g682a0db

Thanks for responding.

timoschilling, 2022-04-23:
We do not support Shelly Dimmer 2! Shelly 2 which you are trying to use is a different device, it’s the pre version of the 2.5

closed 2022-04-23 by timoschilling

timoschilling, 2022-04-25:
> Then wich shelly should I use for my lights?

via @Imburdalinix in https://github.com/mongoose-os-apps/shelly-homekit/pull/978#issuecomment-1107878413

timoschilling, 2022-04-25:
That depends on your needs.
- If you have existing bulbs, then you can use a Shelly 1/1pm/2.5, but then you can't dimm them
- If you would like to dimm and can use 12/24V bulbs use a RGBW2
- If you can replace the bulbs, use Duo/Duo RGBW/Vintage and as switch a I3/Plus I4 with beta 2.11

Imburdalinix, 2022-04-25:
> That depends on your needs.
> 
> - If you have existing bulbs, then you can use a Shelly 1/1pm/2.5, but then you can't dimm them
> 
> - If you would like to dimm and can use 12/24V bulbs use a RGBW2
> 
> - If you can replace the bulbs, use Duo/Duo RGBW/Vintage and as switch a I3/Plus I4 with beta 2.11

Hmm so now I am forced to remove the Shelly Dimmer 2... I technically could do it over homebridge which I will probably do but thank you anyways for the help... I should have informed myself better before buying and intalling this shelly...

andyblac, 2022-04-25:
> Hmm so now I am forced to remove the Shelly Dimmer 2... I technically could do it over homebridge which I will probably do but thank you anyways for the help... I should have informed myself better [...]

you could use Homebridge and the Shelly plugin, that's what I do.