mongoose-os-apps/shelly-homekit #981: Shelly 2.5 has “episodes” of several consecutive reboots

seidnerj, 2022-04-27:
Once every couple of days, one of my Shelly 2.5 that is hooked up to two separate lights gets into these “episodes” where it reboots 4-5 times consecutively. 

This Shelly controls two main living room lights so it is very noticeable when this happens since it results in the lights going off then on again (they are restored to the setting before the reboot).

It can then be seen that the “uptime” statistic has been reset which confirms that a reboot (or similar) has happened. 

I have many Shelly of various types, including other 2.5 in the house all running the same firmware version and non of these display this behavior as far as I can tell.

The log is of course empty and does not display anything that preceded the reboot.

Could this just be some kind of hardware issue?

How can this be debugged or should I just not bother and try to replace with another Shelly?

Thanks in advance

timoschilling, 2022-04-27:
Can you please test your Shelly with 2.10.0, you can download that file here: https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.10.0

seidnerj, 2022-04-27:
Will do. Will update the thread when I have more info.

Thanks!

rojer, 2022-04-28:
also, please check `http://x.x.x.x/debug/core` location. if the device crashes due to software bug, it records a crash dump there. if you get anything, please send it to me directly to rojer@rojer.me

seidnerj, 2022-04-29:
Just observed this happening with 2.10 as well. Thoughts?

seidnerj, 2022-04-30:
I'm starting to think this is due to the temperature of the device, today it essentially gave out for a few minutes. I disconnected the electricity then reconnected it a few minutes later and it started working again.

This specific Shelly is adjacent to another Shelly 2.5, immediately after reconnecting it its temperature read as 86 Celsius (via the web UI). The other adjacent Shelly is only at 76 Celsius and is not exhibiting any a abnormal behavior. I also noticed some of the plastic of the switch adjacent to the troublesome Shelly is looking a bit charred (brownish color).

I suspect all of this behavior is due to overheating.

Is there a way to confirm this? Can I somehow log the temperature level of Shelly overtime (without writing something to continually scrape the UI)?

What is the max operating temperature for the Shelly 2.5 anyways?

Thanks in advance

seidnerj, 2022-04-30:
After upgrading back to 2.10.1, temperature is holding steady at 75 Celsius, even when both Shellys have all lights connected to them powered on. 

This is lower than before but still pretty high. 

Thoughts?

seidnerj, 2022-05-16:
I think I've pretty much confirmed this happens when the temperature rises above 80 something Celsius and stays there for a while. 

Is there any way to log incidences of very high temperature such that the log persists after a unscheduled reboot?

jonasman, 2022-05-22:
i also have one shelly 2.5PM rebooting sometimes but not as often as you. No core dump. I will check temps when this happens.

I have another 2.5PM that just stops working and needs a power cut to come back online.

seidnerj, 2022-06-03:
I've written a small python utility that fetches temperatures (via WebSockets) for all shelly devices discovered on the network (via Bonjour). The device I have failing is always has the highest temperature in the house. Even when both relays are OFF, its temperature holds steady at 60 degrees celsius. Does this make sense?

timoschilling, 2022-06-03:
@seidnerj just for the record: we have a out script in `tools/` in this repo which does the same. Call it simply with `-l -i 3`

seidnerj, 2022-06-03:
Just took a peek, looks almost identical to my code  :)

rojer, 2022-06-03:
2.11.0-beta4 added selector for power saving modes. current default is 0 (no power saving), try setting 1 or 2. it is guaranteed to lower the device temperature but some users reported connectivity issues. please install beta4 and let me know if power saving makes a difference for you - https://github.com/mongoose-os-apps/shelly-homekit/issues/868

seidnerj, 2022-06-03:
Installed it just now, I've set power saving to 1. Will update the thread when I have more information. 

I've been toying with the idea of adding a small cpu fan or heat sink to this specifically Shelly, inside the wall. Hopefully its not going to come to that.

Thanks a lot for your help!

seidnerj, 2022-06-04:
Did not see an improvement in temperature when set to "1", I then tried using "2" and haven't seen a significant difference as well. 

After refreshing the page every couple of minutes over an hour to track the temperature. The Shelly in question started resetting every time I'd hit the status page. 

Finally, after a few of those it became completely unresponsive. Only the manual switches now work. Cannot access the status page and the accessories attached (lights) now show as unresponsive in the Home app.

I will cycle the power later, hopefully it will revive the device.

seidnerj, 2022-06-04:
I cycled the power but the device is still unresponsive, only the manual switches work. Any advice?

seidnerj, 2022-06-04:
I'm going to try device recovery and see if this helps.

seidnerj, 2022-06-04:
I was able to forcefully reset the device by holding the physical button. Back to square one 🤷‍♂️

closed 2022-06-10 by seidnerj

jonasman, 2022-06-10:
So after some days my shelly that was turning off had to be replaced under warranty, this time i took the Plus.
It was a bit too hot around 60 to 70, and maybe the chip didnt handle this temp and died. Power saving modes didnt help on stock firmware either however i did see some lower temps.

my second 2.5PM sometimes reboots the lights no idea why. But it is rare.