mongoose-os-apps/shelly-homekit #982: Shelly 2.5 Dead on Home app

contactodorui, 2022-04-29:
I have 3 shelly 2.5 controlling blinds. All 3 are not responding on the home app.

If i connect to them on their IP adress everything is OK.

I've removed one of them from my Home and tried to Add New Accessory. The home app recognized the QR as a blind but cant connect to the accessory "Accessory Not found".

I'm running all 3 shelly on Version:2.10.1 Build: 20211016-123111/2.10.1-gc8923b3-master

Any help?
Thank Rui

contactodorui, 2022-04-30:
@rojer  Can you help? I got the log from an attempt to add the blind.

thanks in advance

131849109621 shelly_debug.cpp:231    No log file, sending new entries
131856426446 shelly_main.cpp:513     Up 131856.41, HAP 0/0/12 ns 0, RAM: 27708/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:2 lhmd:0
131864426600 shelly_main.cpp:513     Up 131864.41, HAP 0/0/12 ns 0, RAM: 27708/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:2 lhmd:0
131870029235 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131871020596 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131872021988 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131872426746 shelly_main.cpp:513     Up 131872.41, HAP 0/0/12 ns 0, RAM: 27708/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:2 lhmd:0
131873025735 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131874029232 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131876024661 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131877027200 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131878027883 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131879022908 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131880026706 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131880426857 shelly_main.cpp:513     Up 131880.41, HAP 0/0/12 ns 0, RAM: 27708/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:2 lhmd:0
131881028743 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131882031171 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131882200107 mg_rpc.c:306            HAP.Setup via WS_in 192.168.1.185:63951
131885010599 mgos_sys_config.c:232   Loading conf2.json
131885118675 mgos_sys_config.c:174   Saved to conf2.json
131885126431 mgos_homekit_adk_rp:257 Stopping server for reset
131885134929 mgos_mongoose.c:66      New heap free LWM: 18240
131885157591 shelly_main.cpp:337     HAP server state: 2
131885167531 HAPPlatformTCPStrea:105 0x3fff431c 0.0.0.0:9003 HAP connection closed, ns 0/0/12 i 1
131885173426 shelly_main.cpp:337     HAP server state: 0
131885178384 shelly_main.cpp:319     === Destroying accessories
131885202391 mgos_sys_config.c:232   Loading conf2.json
131885343311 mgos_sys_config.c:174   Saved to conf9.json
131885356818 mgos_homekit_adk_rp:268 Resetting server
131885401493 shelly_main.cpp:260     === Creating accessories
131885412842 shelly_hap_window_c:169 WC 1: mp 272.69, mt_ms 21589, cur_pos 100.00
131885522837 shelly_main.cpp:296     === Starting HAP server (ID 0E:63:A9:64:B0:E8, UUID <n/a>, CN 1)
131885621923 HAPPlatformTCPStrea:187 Listening on 9004
131885641874 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131885665507 shelly_main.cpp:337     HAP server state: 1
131885759567 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131886033798 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131887033664 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131888035434 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131889029748 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131890033630 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131891032666 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131891167393 shelly_main.cpp:513     Up 131891.15, HAP 0/0/12 ns 0, RAM: 27716/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
131892038036 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131893039666 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131894043441 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131895044121 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131896043586 mg_rpc.c:306            Shelly.GetInfoExt via WS_in 192.168.1.185:63951
131899167455 shelly_main.cpp:513     Up 131899.15, HAP 0/0/12 ns 0, RAM: 27716/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
131907167393 shelly_main.cpp:513     Up 131907.15, HAP 0/0/12 ns 0, RAM: 27716/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
131915167471 shelly_main.cpp:513     Up 131915.15, HAP 0/0/12 ns 0, RAM: 27716/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
131920866107 mgos_sntp.c:63          SNTP reply from 216.239.35.4: time 1651309616.238475, local 1651309615.040960, delta 1.197515
131923167437 shelly_main.cpp:513     Up 131923.15, HAP 0/0/12 ns 0, RAM: 27720/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
131931167573 shelly_main.cpp:513     Up 131931.15, HAP 0/0/12 ns 0, RAM: 27720/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
131939167256 shelly_main.cpp:513     Up 131939.15, HAP 0/0/12 ns 0, RAM: 27720/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
131947166981 shelly_main.cpp:513     Up 131947.15, HAP 0/0/12 ns 0, RAM: 27720/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
[... cut ...]
132019167013 shelly_main.cpp:513     Up 132019.15, HAP 0/0/12 ns 0, RAM: 27720/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
132027167476 shelly_main.cpp:513     Up 132027.15, HAP 0/0/12 ns 0, RAM: 27720/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
132035167273 shelly_main.cpp:513     Up 132035.15, HAP 0/0/12 ns 0, RAM: 27720/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0
132043167328 shelly_main.cpp:513     Up 132043.15, HAP 0/0/12 ns 0, RAM: 27720/42160; st 50; 4.1: c:1 mp:272.69 mt_ms:21589 cp:100.00 tp:100.00 md:0 lemd:0 lhmd:0

Stundenplan7b, 2022-04-30:
I think i have same problem, but i can't open ip Adress too. Sometimes remove the fuse helps but a few hours Later same problem appears. The Problems only with Shelly 2.5.

MilanBacaCz, 2022-04-30:
I am same experience as you. I am thinking this is not problem with firmware. Even if you flash original firmware back, the issue will be still there. I am thinking the Shelly 2.5 has hw issue. 

S pozdravem,
Milan Bača

Stundenplan7b, 2022-05-01:
Hello i have a new Information. First i remove the fuse. if i go to the ip Adress safari could not Open the web page. If i reload two times the page Open. The shelly Shows activation time about a few seconds. everthing works Fine at homekit. A few minutes later the shelly are not response in HomeKit and also webpage over ip adress dont response. If i reload webpage again dont response. Tried again the follow page was shown and freeze for 5 Minutes.

[image]

closed 2022-06-09 by github-actions[bot]