mongoose-os-apps/shelly-homekit #984: Shelly UNI Alarm On/Off

5xPa, 2022-05-01:
I currently use Alarm Panel output relay to trigger a Uni using your superb firmware have them set as "contact" to trigger HomeKit Automations. Which works brilliantly.  Would it be possible to have an option Hap Type "Alarm On / Off" 

I have had the beta version running with no problems

Thank you all for the tremendous effort, I have made a donation few weeks back in recognition of your efforts

rojer, 2022-05-01:
you mean HAP service "security system"?

5xPa, 2022-05-02:
Hi Rojer, if that is what it is called basically just to indicate alarm (security system) is armed / disarmed. Currently can be motion sensor, contact etc which would be associated with the security system

timoschilling, 2022-05-02:
double of https://github.com/mongoose-os-apps/shelly-homekit/issues/552

closed 2022-05-02 by timoschilling