mongoose-os-apps/shelly-homekit #985: HomeKit FW 2.10.1 and earlier - Rollershutter with Shelly 2.5 on big Window (2x2m) fails to calibrate correctly

Joumann, 2022-05-02:
I Installed 10 Shelly 2.5 with Homekit FW and had no issues at all with calibrating, or use.
In fact, 8 Windows Size 1,3x1,3m and 2 Windows Size 1,8x1,3m.

Just my 2x2m Window with an even stronger motor stops in half while calibrating with the Homekit Firmware and does noch react at all.
I swapped shellys around to check if this one failed, but no change.
Also with the standard shelly Firmware everything works fine with no issue at all.
I guess it is about the power consumption, or the extended time the roller shutter takes to shut the 2m down.

Anyhow, just tried it again and I need to revert to the original shelly firmware again.

Maybe anyone has a "walkaround" how I could implement the big one as well to Homekit.

Thanks in advance for any helpful support!

timoschilling, 2022-05-02:
Please test it with 2.11 beta, see #868 for how to do. We have changed the calibration parameters there.

theontho, 2022-05-12:
I tested 2.10 stable and 2.11 beta3 on my shelly2.5 in roller shutter mode, and I have a similar issue.  I try to calibrate it and it just doesn't calibrate, the relay seems to go into an on/off infinite loop.  I can't get it to open or close properly from the web interface.  

The shelly2.5 does work fine with the default firmware although and the curtain motor works fine  with my aeotec mini roller shutter controller for z-wave.  The curtain motor is a Dooya KT82TS-1.2/14 from 2016

theontho, 2022-05-12:
Here is a log.  My guess because average power is 3W, because your moving curtains horizontally, and your software is expecting much higher numbers, that it gets stuck in a loop.  Is there a way to set power thresholds for calibration?

"movement time: 1.088 s, avg power: 3 W"

[calibration_loop_log.TXT](https://github.com/mongoose-os-apps/shelly-homekit/files/8675047/calibration_loop_log.TXT)

timoschilling, 2022-05-12:
@rojer can you help?

jonnylamb, 2022-05-23:
I had what looks like the same problem with my window.

Looking at your log file, it appears that either the roller shutter was completely open when you calibrated (you should close it) or you should swap the outputs in the settings. I say this because for the **open** calibration the motor is drawing zero power for the first five seconds:

```
362131390 shelly_output.cpp:63    Output 1: off -> on (precal0)
362137471 shelly_hap_window_c:326 WC 1: State: precal0 -> cal0 (10 -> 11)
362473201 shelly_hap_window_c:463 WC 1: P0 = 0.000
363273450 shelly_hap_window_c:463 WC 1: P0 = 1.012
364073091 shelly_hap_window_c:463 WC 1: P0 = 0.000
364873293 shelly_hap_window_c:463 WC 1: P0 = 1.195
365673133 shelly_hap_window_c:463 WC 1: P0 = 0.000
366473104 shelly_hap_window_c:463 WC 1: P0 = 0.000
367173833 shelly_output.cpp:63    Output 1: on -> off (cal0)
```

And I think the next problem is exactly as you said: the threshold for idle motor detection is 5W which is too high for your motor which seems to pull about 3W:
https://github.com/mongoose-os-apps/shelly-homekit/blob/8d0f98d5a6d672f34b0b77f5847cda22546e6751/mos.yml#L78

You should change that `wc.idle_power_thr` setting to something like `2.0` for your motor. I'm not sure how to do that without altering `mos.yml` and reflashing like I did. There must be a way though?

aqzsed123456, 2022-07-23:
J’ai le même soucis voir ma demande d’aide #1041 
merci de nous aider

aqzsed123456, 2022-07-23:
why close my request #1041 I know this is probably the same problem as request #985 but for now no one has given the solution.

Can you help us?

aqzsed123456, 2022-07-24:
will there be a solution or should we give up and find another system to manage the bay windows?

aqzsed123456, 2022-08-11:
Hello

is there anyone to solve this problem?

there is a new shelly, the shelly plus 2PM suitable for rolling shutters.
This may be the solution to our problem but unfortunately it is not yet possible to switch it to Homekit.

thank you.

timoschilling, 2022-08-12:
I get my Shelly Plus 2 PM today, so maybe support will come in the next weeks

aqzsed123456, 2022-08-12:
thank you timoshilling

Can you let me know when you have flashed it homekit?

timoschilling, 2022-08-13:
Simply subscribe on #1013, I will push a prerelease version there

aqzsed123456, 2022-08-13:
thank you timoshilling

aqzsed123456, 2022-09-05:
Hello, always the same problem for me despite the installation of the last update 2.11.2, is it possible to help me? Thanks in advance

aqzsed123456, 2022-09-12:
Hello, is it possible to create an update that solves this problem? If not, what solution is there? Thanks in advance

TheMecha, 2023-03-13:
Same problem here.
It is the only window that gives me problems, the biggest one I have. Every so often I have it stops working, I have to disconnect and reconnect everything and even then I have a hard time configuring it because it restarts every so often. 
I have changed the Shelly and I have put another new one (already 3) and the same thing happens to it, I have come to the conclusion that it is because of the size of the window.

grimmaldus, 2023-04-01:
Did you fix it ? I'm having almost if not the same issue. It gives me either `implausible_power_consumption_in_close_dir` after unsuccessful calibration or `too_few_steps...`.

SHxKM, 2024-10-26:
Has there been any progress with this by anyone here? I'm considering trying this FW strictly because I'm having connectivity issues with some Shellys, but I do have some large windows.

markirb, 2024-10-27:
Hi,

not really. but somebody just posted a similar issue so lets continue here.

I think the underlying problem is that for big windows the consumed Power is higher than for small ones.
This firmware sees everything over 5W (fixed value) as idle power, and I do not think that is correct for big windows.
You could try by setting the shelly to "normal mode" (non window covering) and just drive via switching the output and look at the power in WebInterface.

This variable can be changed but is not accessible on the frontend currently so you can use these command to adjust them and try to recalibrate afterwards. Reboot might be necessary for the change to take effect.

`curl http://a.b.c.d/rpc/Config.Set -d '{"config": {"wc1": {"idle_power_thr": 6.0}}}`
`curl http://a.b.c.d/rpc/Config.Save`

Please try and see....

Nastras, 2024-10-27:
@markirb Hey, thanks for the tip. I think your assumption could be true. Unfortunately, I didn't manage to test it today. I will definitely try it tomorrow or the day after tomorrow and report back with the result.

Nastras, 2024-11-02:
@markirb Hello, please excuse the late reply. Unfortunately I didn't get around to trying it out earlier.

I was able to confirm your assumption that more power than 5W is consumed. I even saw 12W for a short time.
I then adjusted the firmware with the two curl commands you provided. I tried 6W, 12W and 25W including restarting the firmware. Unfortunately, I can't see any change in the original behavior of the control and calibration of the window.
It still only moves in very short sections.
I think we are on the right track to find the error. Do you have any other ideas what I can try?

[image]

markirb, 2024-11-02:
> @markirb Hello, please excuse the late reply. Unfortunately I didn't get around to trying it out earlier.
>
>
>
> I was able to confirm your assumption that more power than 5W is consumed. I even saw 12W [...]

Ok. So you say calibrate only moves shortly? 

Can you please attach logs, then I can see more

Nastras, 2024-11-02:
I have set the firmware to 25W. Here is the calibration log.

112705633 shelly_main.cpp:492     Up 112.69, HAP 0/1/16 ns 1, RAM: 209104/180916; st 35; 4.1: c:1 mp:3.95 ip:25.00 mt_ms:5090 cp:50.86 tp:50.86 md:0 lmd:1
114141728 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
115985155 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
117006817 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
118235888 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
118718090 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.178.25:60599
118740844 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
118804504 shelly_hap_window_c:330 WC 1: State: idle -> precal0 (0 -> 10)
118902612 shelly_hap_window_c:450 Begin calibration
119038614 mgos_sys_config.c:323   Saved to conf9.json
119044141 shelly_output.cpp:67    Output 1: off -> on (precal0)
119050158 shelly_hap_window_c:330 WC 1: State: precal0 -> cal0 (10 -> 11)
119058827 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
119403757 shelly_hap_window_c:466 WC 1: P0 = 5.378
120025006 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
120203626 shelly_hap_window_c:466 WC 1: P0 = 0.000
120705450 shelly_main.cpp:492     Up 120.69, HAP 0/1/16 ns 1, RAM: 209484/180916; st 35; 4.1: c:0 mp:3.95 ip:25.00 mt_ms:5090 cp:50.86 tp:50.86 md:0 lmd:1
120898704 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
121003745 shelly_hap_window_c:466 WC 1: P0 = 1.793
121803661 shelly_hap_window_c:466 WC 1: P0 = 2.872
122603664 shelly_hap_window_c:466 WC 1: P0 = 3.518
123403686 shelly_hap_window_c:466 WC 1: P0 = 4.043
124104465 shelly_output.cpp:67    Output 1: on -> off (cal0)
124110511 shelly_hap_window_c:330 WC 1: State: cal0 -> postcal0 (11 -> 12)
124205154 shelly_hap_window_c:330 WC 1: State: postcal0 -> precal1 (12 -> 13)
124303761 shelly_output.cpp:67    Output 2: off -> on (precal1)
124309704 shelly_hap_window_c:330 WC 1: State: precal1 -> cal1 (13 -> 14)
124903763 shelly_hap_window_c:497 WC 1: P1 = 1.579
125703819 shelly_hap_window_c:497 WC 1: P1 = 1.061
126503700 shelly_hap_window_c:497 WC 1: P1 = 2.744
127303713 shelly_hap_window_c:497 WC 1: P1 = 3.762
128103826 shelly_hap_window_c:497 WC 1: P1 = 4.713
128705475 shelly_main.cpp:492     Up 128.69, HAP 0/1/16 ns 1, RAM: 209484/180916; st 35; 4.1: c:0 mp:3.95 ip:25.00 mt_ms:5090 cp:50.86 tp:50.86 md:0 lmd:1
128903851 shelly_hap_window_c:497 WC 1: P1 = 4.921
129404351 shelly_output.cpp:67    Output 2: on -> off (cal1)
129412066 shelly_hap_window_c:503 WC 1: calibration done, move_time 5089, move_power 3.422
129418033 shelly_hap_window_c:330 WC 1: State: cal1 -> postcal1 (14 -> 15)
129628640 mgos_sys_config.c:323   Saved to conf9.json
129635247 shelly_hap_window_c:350 WC 1: Tgt pos 50.86 -> 50.00 (postcal1)
129641230 shelly_hap_window_c:330 WC 1: State: postcal1 -> idle (15 -> 0)
129651334 shelly_hap_window_c:330 WC 1: State: idle -> move (0 -> 20)
129703539 shelly_output.cpp:67    Output 1: off -> on (move)
129709502 shelly_hap_window_c:330 WC 1: State: move -> rampup (20 -> 22)
129803686 shelly_hap_window_c:552 P = 9.00 -> 3.42
129809779 shelly_hap_window_c:330 WC 1: State: rampup -> moving (22 -> 23)
130405593 shelly_hap_window_c:340 WC 1: Cur pos 9.59 -> 11.55, P = 0.00
131205720 shelly_hap_window_c:340 WC 1: Cur pos 25.31 -> 27.27, P = 2.19
132005693 shelly_hap_window_c:340 WC 1: Cur pos 41.03 -> 42.99, P = 3.06
132405732 shelly_hap_window_c:350 WC 1: Tgt pos 50.00 -> 50.85 (fixup)
132410763 shelly_output.cpp:67    Output 1: on -> off (moving)
132416552 shelly_hap_window_c:330 WC 1: State: moving -> stop (23 -> 24)
132768988 mgos_sys_config.c:323   Saved to conf9.json
132775633 shelly_hap_window_c:330 WC 1: State: stop -> stopping (24 -> 25)
132787265 shelly_hap_window_c:330 WC 1: State: stopping -> idle (25 -> 0)
136705388 shelly_main.cpp:492     Up 136.69, HAP 0/1/16 ns 1, RAM: 209460/180916; st 35; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:50.85 tp:50.85 md:0 lmd:1
144705116 shelly_main.cpp:492     Up 144.69, HAP 0/1/16 ns 1, RAM: 209484/180916; st 35; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:50.85 tp:50.85 md:0 lmd:1

Nastras, 2024-11-02:
Window close:
241134869 shelly_debug.cpp:231    No log file, sending new entries
248705126 shelly_main.cpp:492     Up 248.69, HAP 0/1/16 ns 1, RAM: 209468/180916; st 35; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:50.85 tp:50.85 md:0 lmd:1
256705046 shelly_main.cpp:492     Up 256.69, HAP 0/1/16 ns 1, RAM: 209468/180916; st 35; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:50.85 tp:50.85 md:0 lmd:1
264705442 shelly_main.cpp:492     Up 264.69, HAP 0/1/16 ns 1, RAM: 209468/180916; st 35; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:50.85 tp:50.85 md:0 lmd:1
267335743 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
268358605 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
269382613 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
269997334 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.178.25:60599
270003442 shelly_hap_window_c:350 WC 1: Tgt pos 50.85 -> 0.00 (RPC)
270022597 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
270085392 shelly_hap_window_c:330 WC 1: State: idle -> move (0 -> 20)
270184654 shelly_output.cpp:67    Output 2: off -> on (move)
270190570 shelly_hap_window_c:330 WC 1: State: move -> rampup (20 -> 22)
270274393 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
270287752 shelly_hap_window_c:552 P = 6.54 -> 3.42
270294875 shelly_hap_window_c:330 WC 1: State: rampup -> moving (22 -> 23)
270686846 shelly_hap_window_c:340 WC 1: Cur pos 45.27 -> 43.31, P = 1.67
271431230 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
271486848 shelly_hap_window_c:340 WC 1: Cur pos 29.55 -> 27.59, P = 1.47
272286800 shelly_hap_window_c:340 WC 1: Cur pos 13.83 -> 11.87, P = 2.25
272305076 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
272705191 shelly_main.cpp:492     Up 272.69, HAP 0/1/16 ns 1, RAM: 209472/180916; st 35; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:4.01 tp:0.00 md:2 lmd:2
275385701 shelly_output.cpp:67    Output 2: on -> off (moving)
275391671 shelly_hap_window_c:330 WC 1: State: moving -> stop (23 -> 24)
275562194 mgos_sys_config.c:323   Saved to conf9.json
275568706 shelly_hap_window_c:330 WC 1: State: stop -> stopping (24 -> 25)
275587253 shelly_hap_window_c:330 WC 1: State: stopping -> idle (25 -> 0)
280705284 shelly_main.cpp:492     Up 280.69, HAP 0/1/16 ns 1, RAM: 209472/180916; st 35; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:0.00 tp:0.00 md:0 lmd:2

Nastras, 2024-11-02:
Open Window:
315481299 shelly_debug.cpp:231    No log file, sending new entries
320705264 shelly_main.cpp:492     Up 320.69, HAP 0/1/16 ns 1, RAM: 209488/180916; st 35; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:0.00 tp:0.00 md:0 lmd:2
328705243 shelly_main.cpp:492     Up 328.69, HAP 0/1/16 ns 1, RAM: 209488/180916; st 35; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:0.00 tp:0.00 md:0 lmd:2
336705305 shelly_main.cpp:492     Up 336.69, HAP 0/1/16 ns 1, RAM: 209464/180916; st 34; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:0.00 tp:0.00 md:0 lmd:2
344705264 shelly_main.cpp:492     Up 344.69, HAP 0/1/16 ns 1, RAM: 209488/180916; st 34; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:0.00 tp:0.00 md:0 lmd:2
352705272 shelly_main.cpp:492     Up 352.69, HAP 0/1/16 ns 1, RAM: 209488/180916; st 34; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:0.00 tp:0.00 md:0 lmd:2
360705198 shelly_main.cpp:492     Up 360.69, HAP 0/1/16 ns 1, RAM: 209464/180916; st 34; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:0.00 tp:0.00 md:0 lmd:2
368705367 shelly_main.cpp:492     Up 368.69, HAP 0/1/16 ns 1, RAM: 209488/180916; st 34; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:0.00 tp:0.00 md:0 lmd:2
370978552 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
371377176 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.178.25:60599
371383368 shelly_hap_window_c:350 WC 1: Tgt pos 0.00 -> 100.00 (RPC)
371392882 shelly_hap_window_c:330 WC 1: State: idle -> move (0 -> 20)
371408505 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
371484627 shelly_output.cpp:67    Output 1: off -> on (move)
371490635 shelly_hap_window_c:330 WC 1: State: move -> rampup (20 -> 22)
371584824 shelly_hap_window_c:552 P = 7.38 -> 3.42
371590928 shelly_hap_window_c:330 WC 1: State: rampup -> moving (22 -> 23)
371686606 shelly_hap_window_c:340 WC 1: Cur pos 0.00 -> 1.73, P = 1.86
371721786 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
372486842 shelly_hap_window_c:340 WC 1: Cur pos 15.48 -> 17.45, P = 1.01
372818943 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:60599
373286853 shelly_hap_window_c:340 WC 1: Cur pos 31.20 -> 33.17, P = 1.52
374086771 shelly_hap_window_c:340 WC 1: Cur pos 46.92 -> 48.89, P = 2.68
374886868 shelly_hap_window_c:340 WC 1: Cur pos 62.64 -> 64.61, P = 3.41
375686784 shelly_hap_window_c:340 WC 1: Cur pos 78.37 -> 80.33, P = 4.27
376486797 shelly_hap_window_c:340 WC 1: Cur pos 94.09 -> 96.05, P = 3.98
376685764 shelly_output.cpp:67    Output 1: on -> off (moving)
376691674 shelly_hap_window_c:330 WC 1: State: moving -> stop (23 -> 24)
376707974 shelly_main.cpp:492     Up 376.69, HAP 0/1/16 ns 1, RAM: 206344/180916; st 34; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:100.00 tp:100.00 md:0 lmd:1
376865456 mgos_sys_config.c:323   Saved to conf9.json
376871964 shelly_hap_window_c:330 WC 1: State: stop -> stopping (24 -> 25)
376888014 shelly_hap_window_c:330 WC 1: State: stopping -> idle (25 -> 0)
384705276 shelly_main.cpp:492     Up 384.69, HAP 0/1/16 ns 1, RAM: 209484/180916; st 35; 4.1: c:1 mp:3.42 ip:25.00 mt_ms:5089 cp:100.00 tp:100.00 md:0 lmd:1

markirb, 2024-11-02:
ohh can you adjust max_ramp_up_time_ms (default is 5000) to something bigger?
reset idle power to 5 for now again please

Nastras, 2024-11-02:
@markirb I am currently on the road. I'll try it out as soon as I get back home.

Is the curl command to change the value correct?

curl http://a.b.c.d/rpc/Config.Set -d '{"config": {"wc1": {"max_ramp_up_time_ms": 6000}}}' 

curl http://a.b.c.d/rpc/Config.Save

Nastras, 2024-11-02:
7957795 shelly_debug.cpp:231    No log file, sending new entries
8495960 shelly_main.cpp:492     Up 8.48, HAP 0/1/16 ns 1, RAM: 209216/191844; st 34; 4.1: c:1 mp:3.91 ip:5.00 mt_ms:9089 cp:50.48 tp:50.48 md:0 lmd:0
10386387 mgos_sys_config.c:323   Saved to conf9.json
16496008 shelly_main.cpp:492     Up 16.48, HAP 0/1/16 ns 1, RAM: 209240/191844; st 34; 4.1: c:1 mp:3.91 ip:5.00 mt_ms:9089 cp:50.48 tp:50.48 md:0 lmd:0
24496061 shelly_main.cpp:492     Up 24.48, HAP 0/1/16 ns 1, RAM: 209668/191844; st 34; 4.1: c:1 mp:3.91 ip:5.00 mt_ms:9089 cp:50.48 tp:50.48 md:0 lmd:0
27194257 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
28173536 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
28833126 mg_rpc.c:315            Shelly.SetState via WS_in 192.168.178.25:61725
28849714 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
28896194 shelly_hap_window_c:330 WC 1: State: idle -> precal0 (0 -> 10)
28994273 shelly_hap_window_c:450 Begin calibration
29108644 mgos_sys_config.c:323   Saved to conf9.json
29114198 shelly_output.cpp:67    Output 1: off -> on (precal0)
29120100 shelly_hap_window_c:330 WC 1: State: precal0 -> cal0 (10 -> 11)
29128683 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
29140858 shelly_hap_window_c:466 WC 1: P0 = 3.872
29895422 shelly_hap_window_c:466 WC 1: P0 = 1.006
30065298 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
30695434 shelly_hap_window_c:466 WC 1: P0 = 1.378
31086171 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
31495526 shelly_hap_window_c:466 WC 1: P0 = 2.476
32109922 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
32295272 shelly_hap_window_c:466 WC 1: P0 = 3.134
32495574 shelly_main.cpp:492     Up 32.48, HAP 0/1/16 ns 1, RAM: 209668/191844; st 34; 4.1: c:0 mp:3.91 ip:5.00 mt_ms:9089 cp:50.48 tp:50.48 md:0 lmd:0
33095285 shelly_hap_window_c:466 WC 1: P0 = 3.878
33107929 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
33895430 shelly_hap_window_c:466 WC 1: P0 = 3.726
34164260 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
34695370 shelly_hap_window_c:466 WC 1: P0 = 4.061
35186840 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
35495378 shelly_hap_window_c:466 WC 1: P0 = 0.000
36206089 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
36295528 shelly_hap_window_c:466 WC 1: P0 = 1.073
37025218 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
37095468 shelly_hap_window_c:466 WC 1: P0 = 0.000
37895375 shelly_hap_window_c:466 WC 1: P0 = 1.073
38049097 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
38695310 shelly_hap_window_c:466 WC 1: P0 = 0.000
39074166 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
39495432 shelly_hap_window_c:466 WC 1: P0 = 0.000
40097639 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
40296337 shelly_hap_window_c:466 WC 1: P0 = 1.055
40496105 shelly_main.cpp:492     Up 40.48, HAP 0/1/16 ns 1, RAM: 209668/191844; st 34; 4.1: c:0 mp:3.91 ip:5.00 mt_ms:9089 cp:50.48 tp:50.48 md:0 lmd:0
41095160 shelly_hap_window_c:466 WC 1: P0 = 0.000
41107291 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
41895163 shelly_hap_window_c:466 WC 1: P0 = 0.000
42146592 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
42695451 shelly_hap_window_c:466 WC 1: P0 = 1.055
43169322 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
43495540 shelly_hap_window_c:466 WC 1: P0 = 1.043
44193279 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
44207479 shelly_output.cpp:67    Output 1: on -> off (cal0)
44213588 shelly_hap_window_c:330 WC 1: State: cal0 -> postcal0 (11 -> 12)
44296766 shelly_hap_window_c:330 WC 1: State: postcal0 -> precal1 (12 -> 13)
44395447 shelly_output.cpp:67    Output 2: off -> on (precal1)
44401419 shelly_hap_window_c:330 WC 1: State: precal1 -> cal1 (13 -> 14)
44496314 shelly_hap_window_c:497 WC 1: P1 = 2.299
45217416 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
45295587 shelly_hap_window_c:497 WC 1: P1 = 1.152
46095418 shelly_hap_window_c:497 WC 1: P1 = 1.896
46109188 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
46895433 shelly_hap_window_c:497 WC 1: P1 = 2.445
47266740 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
47695395 shelly_hap_window_c:497 WC 1: P1 = 4.073
48085477 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
48496204 shelly_main.cpp:492     Up 48.48, HAP 0/1/16 ns 1, RAM: 209668/191844; st 35; 4.1: c:0 mp:3.91 ip:5.00 mt_ms:9089 cp:50.48 tp:50.48 md:0 lmd:0
48505251 shelly_hap_window_c:497 WC 1: P1 = 4.921
49109216 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
49295599 shelly_hap_window_c:497 WC 1: P1 = 5.000
50095305 shelly_hap_window_c:497 WC 1: P1 = 5.274
[... 75 lines cut ...]
66197578 shelly_hap_window_c:340 WC 1: Cur pos 20.20 -> 20.85, P = 4.01
66997388 shelly_hap_window_c:340 WC 1: Cur pos 25.43 -> 26.08, P = 4.06
67033032 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
67797382 shelly_hap_window_c:340 WC 1: Cur pos 30.66 -> 31.32, P = 3.84
68155719 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
[... cut ...]
74096408 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
75119560 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
76143080 mg_rpc.c:315            Shelly.GetInfoExt via WS_in 192.168.178.25:61725
80496298 shelly_main.cpp:492     Up 80.48, HAP 0/1/16 ns 1, RAM: 209656/191844; st 35; 4.1: c:1 mp:4.58 ip:5.00 mt_ms:15289 cp:49.63 tp:49.63 md:0 lmd:1
88496130 shelly_main.cpp:492     Up 88.48, HAP 0/1/16 ns 1, RAM: 209624/191844; st 35; 4.1: c:1 mp:4.58 ip:5.00 mt_ms:15289 cp:49.63 tp:49.63 md:0 lmd:1

Nastras, 2024-11-02:
@markirb Hey, I have good news. I set the time to 15000 for testing and now it works as desired. I will stop the actual runtime of the window right away and adjust it again.

Would it be possible for you to include this function in the GUI? 
Thank you very much for your support 👍💪👌

markirb, 2024-11-06:
@Nastras can you find out in detail how much you have to change the value for it to work? 
If it is not much we could even adjust the defaults.

Nastras, 2024-11-06:
I had to set it to 13000 for it to work. It didn't work properly below that.

markirb, 2024-11-15:
ok I fear this is a little bit much for normal windows. I mean you can try on your normal windows - does it still work?

MvonWisotzky, 2025-01-02:
Hi, worked for me as well - but with a value of 21000. Below that the window still stopped before being in end position.

Itsme1609, 2025-03-28:
Where did you change the values as it is not possible to do this in the GUI ?

Itsme1609, 2025-03-28:
Here is the content of the log file

[image]