mongoose-os-apps/shelly-homekit #987: MAC address ERROR in v2.11 beta

lalolerry, 2022-05-04:
Hallo everybody,
I'm using some Shelly devices updated to firmware v2.11 beta3 (a lot of new features in it),
but the MAC addresses shown in the web UI are always slightly different (the second digit) from the ones read with the stock firmware.
There is a little bug in MAC reporting in v2.11?

timoschilling, 2022-05-04:
Can you post some screenshots where we can see that difference? Maybe one from the HomeKit Shelly Ui and one from your router Ui which maybe shows the Mac too

lalolerry, 2022-05-04:
> Can you post some screenshots where we can see that difference? Maybe one from the HomeKit Shelly Ui and one from your router Ui which maybe shows the Mac too

Sure!
Here they are:
[image]
[image]
[image]
[image]

lalolerry, 2022-05-04:
(sorry I closed issue by mistake. Just reopened)

MagOO33, 2022-05-05:
Hello,
i confirm that I have the same error on beta 3 UI that shows a wrong MAC.

rojer, 2022-05-06:
the difference is that we are reporting AP MAC but when the device joins the network it uses its STA MAC.

timoschilling, 2022-05-06:
Ah ok, we should change that. @rojer how can we display the STA Mac?

rojer, 2022-05-10:
i need to add a library method for that, will do

rojer, 2022-05-21:
fixed, will be in the next release

closed 2022-05-21 by rojer