mongoose-os-apps/shelly-homekit #988: Roller Shutter's show 1% less

Imburdalinix, 2022-05-08:
I have shelly's 2.5 installed all over my house. These are controlling all my roller shutters. My problem is that when I (for instance) open my roller shutter for 2% and wait 2 or 3 seconds it will update and just say 1% instead of 2%. This isn't a big deal but if I create a automation that opens my roller shutters for 1% it will show them as closed instead of the 1%. If anyone has any idea of how to fix the problem if would be very helpful. 

If you need my current version here it is:
2.10.1

Thx for the help!

timoschilling, 2022-05-08:
Try to recalibrate them.

Imburdalinix, 2022-05-08:
> Try to recalibrate them. 

I tried to but it didn't work out for me...

timoschilling, 2022-05-08:
@rojer any idea?

Imburdalinix, 2022-05-09:
> @rojer any idea?

I've recorded a screen recording of my home app. I don't know if it helps but maybe it could...

https://user-images.githubusercontent.com/85232768/167343395-101c4286-60ba-497e-bde1-975230372046.MOV

My Home App is set to german... So if you need any help translating i could help you.

Thanks for your help!

rojer, 2022-05-10:
this is because we have a margins of error for positioning, i.e. if we are within 1% of the target we call it done.
things are not perfect in the real world :)

rojer, 2022-05-10:
so this is working as intended.

closed 2022-05-10 by rojer