mongoose-os-apps/shelly-homekit #989: Shelly 1 Without Response on Homekit

KumanduS, 2022-05-08:
i have the same setup for a year: Shelly 1, conected to devolo wireless power line and apple tv. Everything was running well until a month ago that the device start showing "No Response". Based on: https://github.com/mongoose-os-apps/shelly-homekit/wiki/Trouble-Shooting#no-response-or-unable-to-add-accessory-in-homekit
I already:
- Set a new channel to wi-fi;
- Shelly reset factory;
- Re-instaled mongoose on shelly;
- Remove and add the devide;
- Update every software and firmware
But so far i am out of ideias. i dont have any problem adding the device. The iphone even find the device by searching, the problem is that the state of the device does not change it keeps showing "No Response".
Since the devolo and shelly firmware are from 2021 could it be something releated to TvOS (using IPad is not an option)?  
Best regards,

Rui

KumanduS, 2022-05-08:
I have two shelly S. No problem found where. The diference is that they are connect directly (wi-fi) to the router

KumanduS, 2022-05-08:
I have two other shelly (shelly S model) both are showing the same problem. To try to diagnose the problem i made the following setup: 

- Shelly 1 -> Devolo Wi-fi Garage (Id: my web Garage 2.4)-> Router -> Apple TV A connected to Devolo Wi-fi Bedroom (Id: my web Garage 2.4) -> NOK

- Shelly S (WC) -> Devolo Wi-fi Hall Router Wif-fi (Id: my web Garage 2.4) -> Apple TV A connect to Devolo Wi-fi Bedroom (Id: my web Garage 2.4) -> NOK

- Shelly S (Bedroom) -> Wi-fi Bedroom (Id: my web 2.4) -> Apple TV B  connected to Router (Id: my web 2.4)  ->  NOK

Notes: 
- WebPage Shelly 1 - OK (regardless the Id i am connected)
- WebPage Shelly S (WC) -> OK (regardless the Id i am connected)
- WebPage Shelly S (Bedroom) ->OK (regardless the Id i am connected)

Everything was working good until one month ago. My best bet is that some update was made on homekit software or router. Does someone have the same problem?

KumanduS, 2022-05-08:
I updated Shelly 1 to 2.11.0-beta3 and the problem still happens.

KumanduS, 2022-05-09:
I only have this three.

Yes i did. I don't have any problem pairing them to homekit (the homekit can even find them without the QR code). The problem is that after paired they stay in the "without response" mode. I can access the web page of each device. This is a new problem, this set up was working flawless for a year.

Setup:
- Model wifi setup: - my web 2.4 GHz
                               - my web 5 GHz
- Power line wifi setup (DEVOLO Magic 2: 3 nodes, one connected to router, one in the garage, one at home): - my web Garage 2.4 GHz
                                                                                                                                                                                - my web Garage 5 GHz   
- 3x Apple TV: - one connect to "my web 5 GHz"
                        - one connect to "my web 2.4 GHz"
                      - one conected to "my web Garage 2.4 GHz" (this last one i connect yesterday for testing, it was on "my web 2.4 GHz").
 - Shelly: 1 - connected in the garage to "my web Garage 2.4 GHz"
               S- connected in the bedroom to "my web 2.4 GHz"
               S- connected in the WC to "my web Garage 2.4 GHz" (this last one i connect yesterday for testing, it was on "my web 2.4 GHz").
- Aquara hub and devices at the garage - No problem
- Hue Hub and devices at home - No problem
                        
I did not made any network changes it simple stop working. Since the devolo firmware and shelly firmware are from 2021 it may be something related to TvOS or Router firmware (ISP provided by NOS Portugal) update.          

Does any one have an idea?

contactodorui, 2022-05-11:
I have similar problem to you with 3 Shelly 2.5. 

I'm also a NOS client.

Could it be the router?

KumanduS, 2022-05-11:
Hi @contactodorui  my name is also Rui! :D
I really don't know, i did not had any problem until one month ago. Maybe someone can share details of the ports/protocol that are being used after the pairing (the problem is after the pairing).
Details of my router (maybe you can compare the version):
Model HW: CHITA-Hub5
SW version: 7.2.4.7.1b1
App version: 2.81
Unfortunately there is no release date of the software. I did some research and some users are complaing about this version (release date around April): https://forum.zwame.pt/threads/giga-router-hitron-5-0-v2.1051489/page-70
Can some one please specify what is happen after the pairing so i can try to contact the ISD provider?
Best regards,

Rui Nogueira

contactodorui, 2022-05-11:
Same here, thou I can't even pair my 2.5's after the errors. 

Everything was ok until a couple of weeks back.

contactodorui, 2022-05-21:
> Hi @contactodorui  my name is also Rui! :D
>
> I really don't know, i did not had any problem until one month ago. Maybe someone can share details of the ports/protocol that are being used after the [...]

Hi. Where you able to solve this issue?

closed 2022-06-28 by github-actions[bot]

skupu, 2022-11-09:
I have lots of Shelly 1L
Recently I have added 2 new - one was actually some old and with broken rela (I was able to add it to HomeKit after some more time, but possible), so I replaced it from the new one from the shop.
And with this pure new I have a problem.
I cannot add to HomeKit.
It shows the QR, but it is not pairable in iPhone nor possible to be added to HA. iPhone gets timeout, while HA says there is no device waiting for pairing.
What is more, it is not visible also on stock firmware.

How do you think - what might be a reason and what is the solution?