mongoose-os-apps/shelly-homekit #992: control Shelly 2.5 and 1PM flashed for homekit through Dingz

Davidou333, 2022-05-17:
Hello,

I am new and have not much knowledge about all this and I am stuck with my Dingz interruptor (https://www.dingz.ch/).

I was able to put a cmd in the dingz app in order to control my shelly before I flash it. Now this cmd don't work anymore and I would like to know which cmd I can put in order to be able to controle the Shelly trough the Dingz ? 

This was the cmd who was working: Get://192.168.1.37/relay/0?turn=toggle

Thanks a lot for your help!!

rojer, 2022-05-17:
i have no idea what this is about. this project has nothing to do with dingz or its app.

timoschilling, 2022-05-20:
I think his question is how to turn on the relay with via http request.

Davidou333, 2022-05-20:
Thanks a lot to
Reopen it! You’re right I would like to know which command I need
To pu in order to open or close the roller shutter on flashed Shelly 2.5 and same question to toggle light with Shelly 1pm

closed 2022-06-27 by github-actions[bot]