mongoose-os-apps/shelly-homekit #993: Shelly Plug S - Initial State Off: Outlet is ON for about two seconds before being turned off

eilersKevin, 2022-05-19:
In the mode 'Initial State Off' the outlet has power for about two seconds after a reboot before it is turned off again.
This does not happen with the stock firmware.

eilersKevin, 2022-05-19:
I do not, but I just enabled one, to see whether it makes a difference. Unfortunately it did not. 

It seems that the relay is switched on as soon as there is power, and that it then takes 2-4 seconds to reach line 227 in shelly_switch.cpp where it is disabled again. You can clearly hear the click of the relais right after providing power.
This click does not happen with the stock firmware, so it seems that the decision to switch it on or off is taken much earlier there.

eilersKevin, 2022-05-19:
I also tried to play around with the 'Invert Output' option, to maybe trick it into a different behavior, but to no avail.

timoschilling, 2022-05-19:
Can you please post a screen shot of your switch settings and which version are you currently using?

eilersKevin, 2022-05-20:
[image]
[image]

Version: 2.10.1
Build: 20211016-123215/2.10.1-gc8923b3-master 

Homekit is currently disabled but enabling it, does not change the behaviour.

rojer, 2022-05-20:
repro'd this. it's... weird. not sure what sets GPIO15, but something does. i'll have a closer look later.

rojer, 2022-05-21:
a pretty interesting bug, see https://github.com/cesanta/mongoose-os/commit/43e90e0cf30d3b2c9b53f75d79c12e56f7d45b17 for description.

the reason it does not happen in stock fw is that they probably configure GPIO15 (relay) first, then GPIO13 (button), or do it in quick succession such that it's not noticeable.
in our case we have the pause to enter failsafe mode that only configures the button to read its status but due to this bug also drives the relay to on.

thanks for reporting, this will be in the next release.

closed 2022-05-21 by rojer

eilersKevin, 2022-05-22:
Thanks for fixig this so quickly, looking forward to the next release.