mongoose-os-apps/shelly-homekit #996: Shelly Scanner for Shelly Homekit Devices

lalolerry, 2022-05-22:
Hallo everybody,
in my shelly studying and learning, I come across this (IHMO) very usefull tool for Shelly with original firmware.
https://www.usna.it/shellyscanner/index_en.php
having 30+ shelly homekit devices working (perfectly) at my home, this java app would be very usefull for me for their monitoring and management.
I'm not a programmer at all (I wish I were), but maybe someone could "port" (with the help of the original developer) this program to shelly homekit devices. Dreaming?
BR

timoschilling, 2022-05-22:
We have our own script to manage HomeKit Shelly’s. If you want to use that tool, request support there.

closed 2022-05-22 by timoschilling

lalolerry, 2022-05-22:
Uhm.. thanks for good advice, I'll try to contact and ask the developer. But doesn't he need Shelly homekit APIs?
 
BTW, if you refer to our pyton script, it doesn't see to as powerful in managing Shelly Devices as this app. Maybe I'm not able to..
Thanks

timoschilling, 2022-05-22:
Our API is described in the wiki.

lalolerry, 2022-05-22:
ah wow!
Ok, I' ve alrerady wrote to him. Cross fingers :)