mongoose-os-apps/shelly-homekit #998: Double (multi) pulse for Shelly 1 Garage Door

Melcom01, 2022-05-24:
I have my Shelly 1 setup to operate my garage door. All working well on the Shelly side of things but my door needs two presses to open and then two presses to close (same with wired button). 

Whilst I can just press the button twice in the Home app, it makes automation a little more difficult. 

The problem with this is like pressing the button twice - 

First press - nothing but Home says it is opening. 

Second press - door starts to open but Home now says it is closing

I’ve got the reed switch working so after the set time it reverts to the correct open / close status. 

Whilst researching, I’ve found that a lot of door openers need two pulses so hoping to find a setting for that within Mongose or something. 

Is there any way to send a double pulse? If not, please could this be added as a Feature Request?

(Thank you for all your hard work on this - I’m using this on a few Shelly’s and all have been rock solid)

JoeYo37, 2024-04-10:
Did you ever figure out a solution to this? I am having the same issue.

Melcom01, 2024-04-11:
@JoeYo37 Hi, no I didn't but you've got me thinking about it again...

One solution I have thought of is if you have Homebridge, you could use something like this:

https://github.com/nitaybz/homebridge-delay-switch/blob/master/README.md

Setup the auto off to be something like 500ms

You could then have a simple automation that sends a pulse when it is turned on and the same again when it turns off.

This dummy switch would then be your open / close button

Hope this works for you