mongoose-os-apps/shelly-homekit #1001: Eco mode

jonasman, 2022-05-28:
Hi

the stock firmware supports an eco mode that reduces cpu power and sets wifi to low energy.
The advantage is lower power consumption and lower temp. The disadvantage is higher latency. 
this would be a good addition.

My shelly 2.5pm is crashing every 2 days because of high temperature, 70°, so this mode might help.

im actually trying it on stock firmware to see if it helps.

timoschilling, 2022-05-28:
Is coming in 2.11

marcogruhl, 2022-05-29:
I had the same problem (crashing/heat). It seems the capacitor (100μF 16V) is dying. This seems to be a very common problem. It has nothing (directly) to do with the firmware except, generally with lower temperatures the capacitor will last longer, i think.
P.S.: i bought a new 2.5 and will try to repair the old one. the old runs at over 73°, the new about 61°. 
P.P.S.: in the 2.10.1 release there are some heat-improvements relating to the changelog: Shelly 2.5: reconfigure GPIO16 as input to prevent clash with the PM chip's IRQ output. Reduces power consumption and device temperature.

rojer, 2022-05-31:
no, it's not implemented in 2.11 for all models, only for plus.
i tried lowering CPU frequency to 80 and it does nothing to system temperature. modem sleep makes a huge difference but when we tried enabling it people started complaining, and even on plus models [some are still having problems](https://github.com/mongoose-os-apps/shelly-homekit/issues/868#issuecomment-1133976405).
bottom line is: we should make it an option under system settings. no need to bother with cpu freq it seems, just modem sleep 0-1-2 is enough.

closed 2022-06-02 by rojer

rojer, 2022-06-02:
released in 2.11.0-beta4