mongoose-os-apps/shelly-homekit #1002: Faster response

jezdinskyjiri, 2022-06-02:
Hey, is there possibility to improve response of the switch when used as dettached device?

rionshin, 2022-06-02:
What device do you use ? I have no problem with the responsiveness at all. 
especially the new Plus devices are really fast

jezdinskyjiri, 2022-06-02:
I am using Shelli 1 to turn on off Ikea bulb with shortcut. not sure if shelly is the bottleneck...

rionshin, 2022-06-02:
What hardware switch  do you use -push  button or not. How do you configure the Shelly in detached mode as Momentarily or as toggle on/off. 
If you for example use normal hardware switch /not push button/ and  in the configuration set up momentarily - then is slow because it cannot understand is it short or long push.

jezdinskyjiri, 2022-06-02:
I use normal hw switches; shelly input mode is toggle on/off.

rionshin, 2022-06-02:
No way :) it’s not from the Shelly, I have 3 LED lights with Shelly 1 and they turn on immediately , you click the shortcut or the HW switch and lamp is on immediately , you can hear the relay switching less that half of second after the switch is clicked . Do you hear the relay click ?

jezdinskyjiri, 2022-06-02:
There is no click. relay is all the time on. I use the switch to turn on/off smart light bulb...

timoschilling, 2022-06-02:
The long time it takes is a know issue, but it’s not an problem on our side. It’s a problem with HomeKit automations. There are two ways to build them. First a classic one, the other the „convert to shortcut“ one. The last one is really slow.

jezdinskyjiri, 2022-06-02:
Yes thats my case unfortunatelly. I want to have ability to control smart lightbulb with phone and hw switch. Is there a better solution?

closed 2022-07-11 by github-actions[bot]