mongoose-os-apps/shelly-homekit #1007: How to to temporarily disable/lock an accessory?

seidnerj, 2022-06-10:
I would like to be able to "lock" or "disable" certain devices (in my case, blinds) either from manual control (i.e. the actual physical switch) or from the Home app (or both). My kids tend to play with the blinds and I was wondering whether or not there was a way to disable an accessory temporarily. 

I could not find a way to do this, it seems this concept is not available in HomeKit in general. I was wondering if there was a way to do this at least for Shelly Homekit devices?

Thanks in advance

seidnerj, 2022-06-10:
I’m not sure what  you mean by conditional automation?

kloibs, 2022-06-10:
> I’m not sure what you mean by conditional automation?

If the Shelly is in detached mode, you’d have to set an automation in HomeKit to control the roller shutter. You could create a scene, called Parental Controls or whatever, where if enabled, the physical buttons wouldn’t do anything. And if the scene is disabled, the physical switches work as normal.

Not perfect, but it would accomplish most of what you’re trying to implement without any further development on the firmware side of things.

seidnerj, 2022-06-10:
Appreciate the explanation. Got it, that’s an option though I have so many accessories it might be a bit challenging to implement, I would also have to gave a separate “parental control” for each accessory which would many tens of scenes just for this. 

Any way to get the same effect also for control via the Home app?

rojer, 2022-06-11:
at least for the physical aspect of this, we can accommodate by adding a "Disabled" option for input mode.
unlike "Detached", it would not export the inputs as additional homekit accessories and thus would not cause device re-enumeration (and possible loss of configuration associated with that). thus, you could disable physical inputs via web ui and re-enable them when the storm has passed :)
the homekit side of things i think would need to be addressed by apple.

seidnerj, 2022-06-11:
That would be great! Thinking about this some more, one thing that could be done to get a  “HomeKitized” experience is having an optional “companion” accessory for each accessory exposed by a Shelly, that enables/disables the primary accessory. 

That way you could also run automations that enable or disable certain accessories given certain conditions.

Leo-W536, 2022-11-07:
I had the same idea [here](https://github.com/mongoose-os-apps/shelly-homekit/issues/761). 
That's why I'm very excited about the result!

mjorgef, 2023-06-30:
Is this idea possible?
I’m going on Vacations and wanted to leave one shutter disable temporarily, because the window will be open for the Cat, but the shutter closed.

seidnerj, 2024-03-07:
Closing this as project is not maintained anymore.

closed 2024-03-07 by seidnerj