mongoose-os-apps/shelly-homekit #1008: Shelly 2.5 reset spontanously: what to do now?

rjealexander, 2022-06-12:
I own 8 2.5 devices. They were all flashed to this mongoose firmware. Until now I have been very satisfied. 
Yesterday somehow 1 of my Shelly 2.5 devices got reset. Now I can't detect it in homekit anymore but I also can't connect to it through it's ip adress, either I can add it to Shelly app anymore. What to do now?

5xPa, 2022-06-12:
I would try power off for say 30 seconds then reboot.

rjealexander, 2022-06-13:
> I would try power off for say 30 seconds then reboot.

Tx 5xPa, rebooting, that is: pushing the hardware button on the Shelly 2.5 for 10 seconds? Like the original Shelly manual says? Or is there another kind of reboot you have in mind?

5xPa, 2022-06-13:
I was suggesting power off 30 seconds, then power on. I had to do that a while back. Been Ok since