mongoose-os-apps/shelly-homekit #1013: Support for Shelly Plus 2PM

herzkerl, 2022-06-15:
I couldn‘t find any comments regarding the Plus 2PM, is there any news regarding this device?

jonasman, 2022-06-19:
I also need this

Sorbat, 2022-07-15:
is there anything new here? can you already say when an alpha/beta version will be available or is planned?

Thank you very much

1974here, 2022-08-02:
Same here.. longing for a plus2 firmware. 🙂

MagOO33, 2022-08-03:
Hello, Shelly 2.5 is replaced by Shelly 2PM by resellers in my country because 2.5 has too many hardware problems. It would be very nice to have a mongoose support for this new shelly.

firstusing, 2022-08-08:
Same here, could only get "plus 2pm´s" for my covers. 
I would too be happy to have a Firmware for it.

c4ris, 2022-08-28:
a firmware would be nice

AG-dingoant, 2022-08-29:
I just installed a Shelly 2 Plus. I would really like to have the firmware as well :-)

rojer, 2022-08-30:
working on it, stay tuned

aqzsed123456, 2022-09-05:
Hello Rojer, what shelly plus 2 PM will solve problem #985? Thank you very much.

rojer, 2022-09-08:
who knows, maybe if it's to do with power measurement somehow. though Plus2PM uses the same power measurement chip, so it's unlikely.

aqzsed123456, 2022-09-08:
Hello rojer, when I do not flash the shelly 2.5 my two large roller shutters work very well with the shelly application on the other hand when I flash them "HomeKit" the calibration hurts so the roller shutters do not close completely. Is it possible to solve this problem by putting a corrective update on shelly 2.5 or to do it on shelly plus 2PM? So that I know if I have to wait for the update of shelly 2.5 or if I need to buy shelly plus 2PM? Thank you very much

solitpapa, 2022-09-19:
hi, as my 1st from 9 shellys 2.5 after almost 2 years of using starts to die I probably need to start thinking about change to 2PM.
I also would appreciate the introduction of your firmware for that device.

azuma82, 2022-10-02:
Hello, I have also the new +2PM an will use it for optimized my fans system with the netatmo Wetterstation in Applehome. Thanks for you're work on it

andybenichou, 2022-10-16:
Hi, thank you for your work it is amazing !! Sorry to ask but do you think the Shelly Plus 2PM will be supported soon, I am using homebridge now but it is not working well? Thank you!

jonasman, 2022-10-19:
Gosh, my shellys 2.5PMs are all dying and im replacing with the new 2PM plus. Hopefully this could come soon :)

jonasman, 2022-12-11:
Hi!
any update on this? any beta version to try?

Rickth64, 2022-12-28:
Would love to see this

rsalesas, 2023-01-02:
I have a 2.5 that works great for my awning, was going to deploy more for my shutters, but see that the 2PM is not supported. Happy to help testing...

bjoerrrn, 2023-01-07:
looking forward to Shelly Plus 2PM support, too

tcdsantos, 2023-01-09:
Hi!
Is this project alive? Are there any plans to add support for Shelly Plus 2PM ?
Thank you.

luisfernandes88, 2023-01-13:
One more 😀 we need this !

WeisserWolf611, 2023-01-20:
I think there will never come an Homekit-Image. Shelly will support matter and apple will support matter. So why waste time?!

WeisserWolf611, 2023-01-20:
All my 2.5 died, so i replaced all with plus 2PM, but it is not working similar and I cant send it back to Allterco, so I have to wait for matter 🤷🏼‍♂️

luisfernandes88, 2023-01-20:
> I think there will never come an Homekit-Image. Shelly will support matter and apple will support matter. So why waste time?!

Where did you see that info ? Shelly will support matter ?

Rickth64, 2023-01-20:
> > I think there will never come an Homekit-Image. Shelly will support matter and apple will support matter. So why waste time?!
> 
> Where did you see that info ? Shelly will support matter ?

https://www.reddit.com/r/shellycloud/comments/10a8bk7/finally_confirmed_matter_support/

timoschilling, 2023-01-20:
There are still reasons why we will continue to work on this firmware. First matter didn't support all things we do in HomeKit. And how knows that matter support for shelly really comes? They already announced HomeKit support for the plus models and it never happened.

WeisserWolf611, 2023-01-21:
> > I think there will never come an Homekit-Image. Shelly will support matter and apple will support matter. So why waste time?!
> 
> Where did you see that info ? Shelly will support matter ?

Official answer ofrom Allterco:
Thank you for contacting us!
We are working on the Matter support.
But only for Plus and Pro series.

harrisonkohn, 2023-01-21:
Do we have any news on update for this any time soon? Is a unit needed to test?

davidjbradshaw, 2023-01-23:
@harrisonkohn the other week they said June/July.

timoschilling, 2023-01-23:
Please don’t discuss over matter support for the stock firmware here, that’s offtopic

timoschilling, 2023-01-27:
I made some progress, but now I bricked my dev Plus2PM :(

5xPa, 2023-01-27:
I would be happy to donate £25 to permit you to purchase a replacement. If you provide a link

Rickth64, 2023-01-27:
I was about to say the same. Let us know so you have a few to brick :-)

timoschilling, 2023-01-27:
https://paypal.me/timoschilling

timoschilling, 2023-01-27:
That would be nice https://paypal.me/timoschilling

5xPa, 2023-01-27:
Timo have made the donation a small token of my appreciation of the work on
the project.
If for any reason the funds do not show up please let me know

1974here, 2023-01-27:
+1Thanks for working on this project, Timo!!//HeikoVon meinem iPhone gesendet

+1

duskiez, 2023-01-30:
need this as well and will send a donation - thanks for working on this

timoschilling, 2023-02-03:
Short update: I get HomeKit working on the 2PM. But now I burned my flasher and one of my Mac Book USB ports 🤦‍♂️. Open points are revertible to stock and power measurement.

jonasman, 2023-02-03:
How did you burn the usb??

duskiez, 2023-02-03:
> Short update: I get HomeKit working on the 2PM. But now I burned my flasher and one of my Mac Book USB ports 🤦‍♂️. Open points are revertible to stock and power measurement.

Awesome, can’t wait for the release!

rsalesas, 2023-02-04:
Really sorry you lost the port - is it repairable? Happy to contribute to compensate.

timoschilling, 2023-02-08:
I was testing with the power measurement, so the Shelly was connected to main power and I miss wired my flasher, I forgot to plug the voltage isolator between the flasher and the Shelly. So the 230V from main went into the flasher. That killed the Shelly, the flasher and one of my USB ports. I think repeating would mean to replace the main board.

rsalesas, 2023-02-08:
I don't think my contribution will cover that unfortunately! 😔

Ioan-Popovici, 2023-02-13:
> I was testing with the power measurement, so the Shelly was connected to main power and I miss wired my flasher, I forgot to plug the voltage isolator between the flasher and the Shelly. So the 230V [...]

Unfortunate, but you should be able to replace the burned controller and port at a shop that still does hardware repair. Don't throw away the board yet ;)

Try to see if you can find someone in your area, it should be pretty cheap.

timoschilling, 2023-02-16:
I worked on it, switch and garage door worked fine, but revertible to stock and power measurement are still open issues.

rsalesas, 2023-02-16:
I would take those restrictions! Nice work!!

rsalesas, 2023-02-16:
Not sure how else to send this but... I would definable buy all my Shelley's on Amazon through an associate link. Would compensate you for all the amazing work.

Captain-Chaos13, 2023-03-05:
@timoschilling: i‘m waiting too😉 hope you are successful!

smuel09, 2023-03-19:
Hi @timoschilling Thank you for your hard work.
Is there an option to install other Shelly firmware on the plus2PM for the time being or it doesn't work like that?

antoine4livre, 2023-04-21:
+1 It would be amazing ! 🙏

sebr5884, 2023-04-21:
Any updates on this? Is a firmware available now?

failbit, 2023-05-03:
Would really love this aswell!!

TheBigNeo, 2023-05-05:
I already had to replace 2 Shelly 2.5 this month. Yesterday the third died and now the 2.5 are no longer available. :(

Sorbat, 2023-05-05:
> I already had to replace 2 Shelly 2.5 this month. Yesterday the third died and now the 2.5 are no longer available. :(
I found some on amazon.fr last week https://www.amazon.it/dp/B08YNVSBSS/

Sorbat, 2023-05-05:
> > I already had to replace 2 Shelly 2.5 this month. Yesterday the third died and now the 2.5 are no longer available. :(
I found some on amazon.fr last week https://www.amazon.it/dp/B08YNVSBSS/

tillmanz-valooba, 2023-05-16:
Hi there!
Just checking in on how things are progressing. :-) 
Apart from my financial contribution is there anything additionally that I could possibly do to support @timoschilling or @rojer with this?

CincoD2, 2023-06-02:
Also waiting this, @rojer ! I'd contribute to get it!

Tillman-Z, 2023-06-02:
Hi @timoschilling 😀 It would be great if you could share some info on the current state of the Plus 2PM support. Are you still working on it?

sebsi1, 2023-06-03:
Need it too :/

[71 more replies]

closed 2024-07-22 by markirb