mongoose-os-apps/shelly-homekit #1014: Option : Connect to cloud

livb33, 2022-06-19:
Hi Dear,

The option "Connect to cloud" is not available on your firmware. On the original one I had this option available. Will this option be developed ? 
Now Im able to use the shelly when I'm connected to the wifi and when I'm connected to the 4G. 
Is there any way to disable this cloud connection ? 

Thank you so much,

Best regards

timoschilling, 2022-06-19:
Our firmware doesn't work with the shelly cloud and that will not be changed in the future.

If you don't want that you can use out firmware remotely, you need to disable that in HomeKit settings.

closed 2022-06-19 by timoschilling

livb33, 2022-06-19:
hi 
thanks for your quick answer. If we activate the cloud access in homekit (setting up hub with Apple TV, homepod or ipad) by which server do we go? only the apple cloud?

timoschilling, 2022-06-19:
Yes in HomeKit the accessories are only talking with your HomeHub (Apple TV, HomePod, iPad) and that one Proxies all data to Apple. Remote access is a HomeKit feature not a accessory feature

livb33, 2022-06-19:
OK thanks a lot.