mongoose-os-apps/shelly-homekit #1015: Shelly 2.5 doing the opposite - FW 2.11.0

00x7, 2022-06-20:
After flashing firmware 2.11.0 all my Shelly 2.5s show some unexpected behaviour to certain Siri command sequences where they end up doing the exact opposite of the command issued.

My general settings:
---
Shelly Mode: Roller Shutter
Input mode: Separate - toggle
Swap Inputs: OFF
Swap Outputs: OFF
Calibrated: YES
Starting state: Shutters fully open
State in Home App: Open

The unexpected behaviour happens in STEP 3, where the command to open triggers a close. When issued again in STEP 4 the same command then opens the shutters correctly. Could others please verify?

STEP 1:
Siri HomeKit command: "Set all shades to 60%"
Siri confirmation: "OK, I set all the shades to 60%."
Result: Shelly lowers the roller shutters from fully open to 60% open
Status in Home App: Closing... -> 60% Open

STEP 2:
Siri HomeKit command: "Set all shades to 70%"
Siri confirmation: "OK, I set all the shades to 60%."
Result: Shelly raises the roller shutters from 60% to 70% open
Status in Home App:	Openinging... -> 69% Open

STEP 3:
Siri HomeKit command: "Open all shades"
Siri confirmation: "OK, all the shades are open."
Result: Shelly lowers shades to fully closed
Status in Home App: Opening... -> Closing... -> Closed

STEP 4:
Siri HomeKit command: "Open all shades"
Siri confirmation: "OK, all the shades are open."
Result: Shelly raises shades to fully open
Status in Home App: Opening... -> Open

Other observations:
Starting state: Open -> Set to 60% -> Set to 70% -> Set to 100% : instead of opening, it fully closes
Starting state: Open -> Set to 60% -> Set to 70% -> Set to 99% : works
Starting state: Open -> Set to 70% -> Set to 60% -> Open : works
Starting state: Open -> Set to 70% -> Set to 60% -> Close: instead of closing, it fully opens

timoschilling, 2022-06-20:
Does the same happens when you give the commands via home app?

00x7, 2022-06-21:
> Does the same happens when you give the commands via home app?

Yes, when I swipe the slider from a fully open state to (1) a partially opened state and then to (2) a slightly more open state and finally to (3) a fully open state, the third action will result in the shades fully closing.

Also, as via Siri, replacing (3) with a position just shy of fully open will work just fine.

Also, and maybe entirely unrelated, after running a time triggered automation taking the shutters from their previous state of fully closed to 60% open, the Home App continues to show all shutters as "Opening..." with a spinning wheel until the next manual interaction is executed.

patdiez, 2022-06-21:
Similar phenomenon with me. If I swipe the slider all the way up, the shutter is closed. However, Siri via voice command works as far as I have been able to test it so far.

https://user-images.githubusercontent.com/52236155/174747288-f20d9e26-ef3a-41a3-b561-1b1bdef5ec78.mov

and the logs:

`913243051 shelly_main.cpp:443     Up 913.23, HAP 0/1/12 ns 1, RAM: 26644/20624; st 51; 4.1: c:1 mp:92.67 ip:5.00 mt_ms:19490 cp:0.96 tp:0.96 md:0 lmd:1
916253235 shelly_hap_window_c:371 WC 1: HAPSetTgtPos 100.00 cur 0.96 tgt 0.96 lmd 1
916259480 shelly_hap_window_c:346 WC 1: Tgt pos 0.96 -> 0.00 (HAPalt)
916265643 shelly_hap_window_c:326 WC 1: State: idle -> move (0 -> 20)
916282359 shelly_output.cpp:63    Output 2: off -> on (move)
916288913 shelly_hap_window_c:326 WC 1: State: move -> rampup (20 -> 22)
916382721 shelly_hap_window_c:549 P = 0.00 -> 92.67
916482688 shelly_hap_window_c:549 P = 0.00 -> 92.67
916583023 shelly_hap_window_c:549 P = 0.00 -> 92.67
916682657 shelly_hap_window_c:549 P = 0.00 -> 92.67
916782763 shelly_hap_window_c:549 P = 0.00 -> 92.67
916882540 shelly_hap_window_c:549 P = 0.00 -> 92.67
916982782 shelly_hap_window_c:549 P = 0.00 -> 92.67
917082633 shelly_hap_window_c:549 P = 0.00 -> 92.67
917182881 shelly_hap_window_c:549 P = 1.01 -> 92.67
917282743 shelly_hap_window_c:549 P = 1.02 -> 92.67
917383050 shelly_hap_window_c:549 P = 0.00 -> 92.67
917482843 shelly_hap_window_c:549 P = 0.00 -> 92.67
917582656 shelly_hap_window_c:549 P = 1.01 -> 92.67
917682862 shelly_hap_window_c:549 P = 0.00 -> 92.67
917783123 shelly_hap_window_c:549 P = 0.00 -> 92.67
917882994 shelly_hap_window_c:549 P = 0.00 -> 92.67
917983038 shelly_hap_window_c:549 P = 0.00 -> 92.67
918082838 shelly_hap_window_c:549 P = 0.00 -> 92.67
918183155 shelly_hap_window_c:549 P = 0.00 -> 92.67
918282642 shelly_hap_window_c:549 P = 0.00 -> 92.67
918382705 shelly_hap_window_c:549 P = 1.02 -> 92.67
918482500 shelly_hap_window_c:549 P = 0.00 -> 92.67
918582493 shelly_hap_window_c:549 P = 1.02 -> 92.67
918682633 shelly_hap_window_c:549 P = 0.00 -> 92.67
918785443 shelly_hap_window_c:549 P = 1.05 -> 92.67
918882780 shelly_hap_window_c:549 P = 1.01 -> 92.67
918982995 shelly_hap_window_c:549 P = 0.00 -> 92.67
919082555 shelly_hap_window_c:549 P = 1.02 -> 92.67
919182545 shelly_hap_window_c:549 P = 0.00 -> 92.67
919282994 shelly_hap_window_c:549 P = 0.00 -> 92.67
919382680 shelly_hap_window_c:549 P = 0.00 -> 92.67
919482674 shelly_hap_window_c:549 P = 0.00 -> 92.67
919583435 shelly_hap_window_c:549 P = 0.00 -> 92.67
919682029 shelly_hap_window_c:549 P = 0.00 -> 92.67
919782493 shelly_hap_window_c:549 P = 1.02 -> 92.67
919882714 shelly_hap_window_c:549 P = 1.02 -> 92.67
919982739 shelly_hap_window_c:549 P = 0.00 -> 92.67
920082499 shelly_hap_window_c:549 P = 0.00 -> 92.67
920182545 shelly_hap_window_c:549 P = 0.00 -> 92.67
920282473 shelly_hap_window_c:549 P = 0.00 -> 92.67
920382634 shelly_hap_window_c:549 P = 1.01 -> 92.67
920467981 shelly_hap_window_c:371 WC 1: HAPSetTgtPos 96.00 cur 0.96 tgt 0.00 lmd 2
920473941 shelly_hap_window_c:346 WC 1: Tgt pos 0.00 -> 96.00 (HAP)
920479713 shelly_hap_window_c:549 P = 1.01 -> 92.67
920496037 shelly_hap_window_c:549 P = 1.02 -> 92.67
920582632 shelly_hap_window_c:549 P = 0.00 -> 92.67
920622256 shelly_hap_window_c:371 WC 1: HAPSetTgtPos 100.00 cur 0.96 tgt 96.00 lmd 2
920628495 shelly_hap_window_c:346 WC 1: Tgt pos 96.00 -> 0.00 (HAPalt)
920634378 shelly_hap_window_c:549 P = 0.00 -> 92.67
920682650 shelly_hap_window_c:549 P = 0.00 -> 92.67
920783104 shelly_hap_window_c:549 P = 0.00 -> 92.67
920882583 shelly_hap_window_c:549 P = 0.00 -> 92.67
920982641 shelly_hap_window_c:549 P = 1.03 -> 92.67
921083591 shelly_hap_window_c:549 P = 1.06 -> 92.67
921182645 shelly_hap_window_c:549 P = 0.00 -> 92.67
921243277 shelly_main.cpp:443     Up 921.23, HAP 0/1/12 ns 1, RAM: 26592/20624; st 51; 4.1: c:1 mp:92.67 ip:5.00 mt_ms:19490 cp:0.96 tp:0.00 md:2 lmd:2
921282713 shelly_hap_window_c:549 P = 0.00 -> 92.67
921383137 shelly_hap_window_c:549 P = 0.00 -> 92.67
921387907 shelly_hap_window_c:556 Failed to start moving
921393873 shelly_hap_window_c:326 WC 1: State: rampup -> stop (22 -> 24)
921482154 shelly_output.cpp:63    Output 2: on -> off (stop)
921644644 mgos_sys_config.c:323   Saved to conf9.json
921652526 shelly_hap_window_c:326 WC 1: State: stop -> stopping (24 -> 25)
921674158 shelly_hap_window_c:326 WC 1: State: stopping -> idle (25 -> 0)
921682669 shelly_hap_window_c:326 WC 1: State: idle -> error (0 -> 100)
921785088 shelly_hap_window_c:346 WC 1: Tgt pos 0.00 -> 0.96 (error)
921791601 shelly_hap_window_c:326 WC 1: State: error -> idle (100 -> 0)
929243221 shelly_main.cpp:443     Up 929.23, HAP 0/1/12 ns 1, RAM: 26644/20624; st 51; 4.1: c:1 mp:92.67 ip:5.00 mt_ms:19490 cp:0.96 tp:0.96 md:0 lmd:2`

marcus-aa, 2022-06-23:
I have the same or similar also in 2.1.10, if it is moving and is ordered in different direction, it stops. 

Moving to 0, is at 21.70, asked to go to 100. Stopping.

This one seems to be the problem:
347 WC 1: Tgt pos 0.00 -> 20.70 (HAPalt)

It should be 20.70 -> 100 ???

```
61068499187 shelly_hap_window_c:599 Moving to 0, p 286.60
61069070526 shelly_hap_window_c:370 WC 1: HAPSetTgtPos 100.00 cur 21.70 tgt 0.00 lhmd 2
61069076817 shelly_hap_window_c:347 WC 1: Tgt pos 0.00 -> 20.70 (HAPalt)
61069084252 shelly_hap_window_c:336 WC 1: Cur pos 21.70 -> 21.35, P = 286.12
61069198917 shelly_hap_window_c:347 WC 1: Tgt pos 20.70 -> 20.88 (fixup)
61069204442 shelly_output.cpp:63    Output 2: on -> off (moving)
61069210286 shelly_hap_window_c:327 WC 1: State: moving -> stop (23 -> 24)
61069316396 mgos_sys_config.c:232   Loading conf2.json
61069455472 mgos_sys_config.c:174   Saved to conf9.json
61069463501 shelly_hap_window_c:327 WC 1: State: stop -> stopping (24 -> 25)
61069475970 shelly_hap_window_c:327 WC 1: State: stopping -> idle (25 -> 0)
```

MagOO33, 2022-08-11:
Hello, same for me.

MagOO33, 2022-09-20:
I think that the buggy thing is homekit real time position report when you put a percentage command.
you can see the roller shutter move in HomeKit.
And if you touch the screen, the current position is a new command.
really a problem.

andybenichou, 2022-10-04:
Hi, I have the same issue, is there a fix for this?

timoschilling, 2023-01-16:
That's what was described in the initial issue is a common problem with Siri. if you use Siri to control a window cover it asks HomeKit for the current state and sets a new state in HomeKit which sometimes is wrong. I have similar issues with my official certificated Velux window covers as well.

closed 2023-01-16 by timoschilling