mongoose-os-apps/shelly-homekit #1016: Shelly UNI doesn't pair

timoschilling, 2022-06-21:
The Shelly UNI doesn't let it pair with HomeKit. I have tested it with two different Shelly UNIs, in GDO and Switch configuration.

@rojer I have already tried to debug it and didn't find any solution, please can you take a look on that. I think you can reproduce it by using a generic dev esp8266 board.

solitpapa, 2022-07-03:
i've got the same issue, but with shelly 1. checked if version may be the reason but no. after downgrade to 2.9.5. also cannot add shelly to homekit. anything in homekit changed?
have many devices added long time ago, updated now to current version 2.11.1 and all working fine but I'm not able to add new. tested with 2 different shellys 1

timoschilling, 2022-07-08:
Problem solved with 2.11.1