mongoose-os-apps/shelly-homekit #1017: Shelly 1 Garagedoor Movement Time is wrong.

PhilippBLG, 2022-06-21:
Heey,

So I updated to Version 2.11 today and found that when the garagedoor is closed and I open it it shorty writes opening for like 5 seconds and then says the Garagendoor is closed and just after the 30seconds when the sensor senses that the door is open it says open. I set the movement time to 45 seconds but it somehow doesn’t use that setting and I always get 2 notifications when opening the garagedoor.

hope there’s a fix for that.

rojer, 2022-06-21:
can you open the debug log page from the web ui, perform the action you describe and post the log here? it should help me understand what's going on.

PhilippBLG, 2022-06-22:
[image]

i think you can see the problem here pretty clearly.

marcus-aa, 2022-06-24:
Maybe this was closed too soon, I get a sense it may be related

https://github.com/mongoose-os-apps/shelly-homekit/issues/609

joddye2, 2022-06-25:
> Maybe this was closed too soon, I get a sense it may be related
> 
> #609

Yes, this problem persist since 2021 (for me)

rameezk, 2022-07-11:
I’ve just stumbled onto this exact issue now too. 

I’ve also tried moving the magnet (of the sensor) and changed between Normally Open and Normally Closed (even swapped between OC and NC switches). But the issue still persists. 

It’s almost as if the “new state” is read at 5 seconds no matter what.

closed 2022-08-18 by github-actions[bot]

marcus-aa, 2022-08-18:
Not completed, should stay open.