mongoose-os-apps/shelly-homekit #1020: Shelly Pro 2 firmware support?

moranchoj, 2022-06-24:
Shelly Pro 2 has already been relased and it is available for selling.

are there plans to develop a firmware version for Shelly Pro 2 with the curtain/shades option?

Shelly Pro 2 has 2 dry contacts and it is very suitable to be used for curtains motors that have 2 control lines not energized with the current of the motor. The curtain motor provides a common line and 2 control lines to activate up and down movement.
 
Do you need a test device to develop the firmware version?

Thanks

timoschilling, 2022-06-25:
Double of #861

closed 2022-06-25 by timoschilling

moranchoj, 2022-06-25:
Why do you close?

Previous plans request for Pro 2 device was done 6 month ago and 1 Plus device is already supported.

I can provide you a Shelly 2 Pro device to do the development. This is the unique model with 2 dry contacts. No one else has this feature and this will enable remote control of some curtain/shades motors with 2 separate circuits for power and control

Thanks

timoschilling, 2022-06-25:
I have closed this because #861 requested already support for all kinds of pro devices, including the pro 2 and we do not accept doubled issues.

If you would like to provide a dev devices you can contact me via email which is written on my profile page https://github.com/timoschilling.