mongoose-os-apps/shelly-homekit #1022: Power saving mode

askarkurymbayev, 2022-06-27:
Good afternoon! I have a question about power saving mode?
There are three kinds of mode: disabled, 1 and 2
what is the difference between mode one and mode two?
And what is the algorithm of these modes?

markirb, 2022-07-16:
The behavior of this is different for ESP8266 (gen 1 shellys) and ESP32 (gen 2 shellys):

ESP8266:
https://github.com/mongoose-os-libs/wifi/blob/master/mos.yml#L88
WiFi sleep mode: 0 - none, 1 - light, 2 - modem
Explanation:
https://blog.creations.de/?p=149

ESP32:
https://github.com/mongoose-os-libs/wifi/blob/master/mos.yml#L103
Power save mode for station: 0 - none, 1 - min, 2 - max.
Explanation: ?

rojer, 2022-07-17:
my understanding is that 1 (light sleep) is a deeper sleep than 2 (modem), the latter only affecting the radio power. but i have not actually measured the power consumption.

closed 2022-08-24 by github-actions[bot]

lucaspinelli85, 2022-10-01:
I still haven’t understood the difference between level 1 and 2.

Can someone explain it to me in a simple and understandable way? Thank you

1974here, 2022-10-02:
Hi, same question here. I did follow the link from markirb but this does not answer the question... At least I do not understand to. Pls help..

markirb, 2022-10-03:
Can you specify more what you don't understand after reading the link?

lucaspinelli85, 2022-10-05:
Currently I still don’t understand anything about the difference between the two modes

markirb, 2022-10-05:
For Gen1 Shelly Devices:
2 - Modem Sleep Mode: your modem (Wi-Fi) will periodically go to sleep i.e. power save mode in between. This saves some energy

1 - Light sleep mode: additionally to the modem the processor will periodically also sleep. This saves even more energy. 

Are there inherent problems? Maybe, depends on configuration and device. light sleep will not work with e.g. ShellyRGBW2 as the PWM of the lights use timers which do not work anymore then.

lucaspinelli85, 2022-10-05:
So level 1 saves more energy than level 2?

Can you tell me what the generation 1 and generation 2 shellys are?

1974here, 2022-10-05:
Thank you Markus, this explanation helps me.. One more question though: If the modem periodically goes to sleep then it will no longer react to Wi-Fi commands. So it is only for users who work with hardwired switches only?

aonez, 2025-06-05:
Here are the explanations:
https://github.com/mongoose-os-apps/shelly-homekit/wiki/WiFi-Settings

> On Plus and newer devices / ESP32 and ESP32C3:
> 
> 2 - Maximal Saving 1 - Minimal Saving 0 - No saving
> 
> On Gen1 Devices / ESP8266
> 
> 2 - Modem Sleep Mode: your modem (Wi-Fi) will periodically go to sleep i.e. power save mode in between. This saves some energy.
> 1 - Light Sleep mode: additionally to the modem the processor will periodically also sleep. This saves even more energy. 0 - No sleep