mongoose-os-apps/shelly-homekit #1023: double-ping

ACWAKKERMANS, 2022-07-03:
Hi there,

I use my Shellys as motion sensors.

I also use HomeBridge on which I use the "HomeBridge Security System" plug-in.
The plug-in has a double-knock feature, requiring sensors to ping twice within x-amount of seconds to trigger the alarm. This cuts down on false alarms as, for example, a bird passing in front of the PIR will only "knock" the once. Conversely, a person passing in front will trigger sensor A (1st knock), go around the corner and trigger sensor B, providing the second knock.

To identify within HomeBridge which sensor actually triggers, I have assigned each Shelly a virtual momentary switch. These virtual switches don't perform any actions, they're indicators only.

Recently I have started seeing my alarm being triggered a lot more. Looking at the HomeBridge logs, I see that the sensor triggering somehow sends two "on" signals in quick succession:

`[7/3/2022, 7:29:20 AM] [Farmers Lane Motion] Setting switch to true
[7/3/2022, 7:29:20 AM] [House Security] Siren Switch (Knock)
[7/3/2022, 7:29:20 AM] [Farmers Lane Motion] Setting switch to false
[7/3/2022, 7:29:21 AM] [Farmers Lane Motion] Setting switch to true
[7/3/2022, 7:29:21 AM] [House Security] Siren Switch (On)
[7/3/2022, 7:29:21 AM] [Farmers Lane Motion] Setting switch to false`

As you can see, HomeBridge receives a second "On" from the one sensor within a second of the first "On"
That shouldn't be possible as the physical PIR's remain "on" for 7-9 seconds as is common.

I am trying to identify if the issue is the Shelly firmware sending two signals in rapid succession or if there's something in HomeBridge causing the problem.

I went into the debug log on one of the Shelleys and then walked in front of the PIR to trigger it:

`489291173596 shelly_main.cpp:443     Up 489291.16, HAP 0/4/12 ns 4, RAM: 25920/19032; st 0; 0.1: st:0 in_st:0 inm:3 ininv:0; 7.1: svt:7 inv:0 st:0 lea:1814.412
489296781777 shelly_input.cpp:76     Input 1: change (state 1)
489296786498 shelly_hap_sensor_b:141 Sensor state: 0 -> 1
489297795258 shelly_input.cpp:76     Input 1: long (state 1)
489299173197 shelly_main.cpp:443     Up 489299.16, HAP 0/4/12 ns 4, RAM: 25920/19032; st 0; 0.1: st:0 in_st:1 inm:3 ininv:0; 7.1: svt:7 inv:0 st:1 lea:2.372
489307172949 shelly_main.cpp:443     Up 489307.16, HAP 0/4/12 ns 4, RAM: 25920/19032; st 0; 0.1: st:0 in_st:1 inm:3 ininv:0; 7.1: svt:7 inv:0 st:1 lea:10.371
489307351425 shelly_input.cpp:76     Input 1: change (state 0)
489307356118 shelly_hap_sensor_b:141 Sensor state: 1 -> 0
489315172967 shelly_main.cpp:443     Up 489315.16, HAP 0/4/12 ns 4, RAM: 25920/19032; st 0; 0.1: st:0 in_st:0 inm:3 ininv:0; 7.1: svt:7 inv:0 st:0 lea:18.371`

I don't know how to interpret the log so was hoping one of you might be able to see if there's two "On" signals being sent or just the one?

Any help would be greatly appreciated!

timoschilling, 2022-07-08:
Yes I thing there there is a bug too. The sensor interpreted the log contact from the motion sensor as long push (which is a logic for the stateless input) and thats bullshit for the motion sensor. It looks like all or at least most of the sensors have that bug.

Maybe a workaround can be that you set your motion sensor to pulse and adjust the Idle Time to your needs.

ACWAKKERMANS, 2022-07-12:
Hmm, thank you for having a look at this! It’s worked fine for 1.5 yrs though.
Anyway, I’m trying your workaround on two of the sensors with an idle time of 9 seconds.

Can I assume that when a physical PIR trigger is received by Shelly on the SW, it’ll  send “ON” to HomeKit for those 9 seconds? What happens if the PIR triggers again after 8 seconds or indeed… remains triggered for longer than 9 seconds… will the Shelly remain triggered then too? Will the 9 second idle time countdown restart?

thanks again for everything!

ACWAKKERMANS, 2022-07-13:
I have tested this work-around.
It seems to stop the "double-ping" issue and works well to prevent my security system triggering.
I also use these PIRs to trigger security lighting. For that this work-around works a little less well. When movement is detected, the security light will correctly come on. However, if movement persists or the PIR is triggered within the idle-time I set (9 sec), the Shelly will not continue to remain "ON" and the light will go off, until the PIR goes "off" and is then triggered again.

I'll report back with any other findings over the coming weeks.