mongoose-os-apps/shelly-homekit #1024: Garage door opener initial state

d4rkmen, 2022-07-03:
Thanks for a graat project first. It's realy something! Shelly devices also are cool

The issue shows afre a power cycle. And actual `closed` gates shows `open`
After very first target state change it sets according to a sensor value.

May be it's related to this line, where the current state characteristic refers to an uninited value `cur_state_`
https://github.com/mongoose-os-apps/shelly-homekit/blob/da78880591d5f2beb62c5d42609796c283b69659/src/shelly_hap_garage_door_opener.cpp#L66

And the actual init happen later here:
https://github.com/mongoose-os-apps/shelly-homekit/blob/da78880591d5f2beb62c5d42609796c283b69659/src/shelly_hap_garage_door_opener.cpp#L98

Thanks in advance

d4rkmen, 2022-07-04:
It seems the same problem with `tgt_state_` 👀

d4rkmen, 2023-04-17:
Its sad, but the issue is still present.
I have made some researches. The close sensor of Shelly1 has "normally closed" default config. And by default all works as expected. But set it to "normally open" and restart to reproduce the issue. 
After restart the closed sensor state will be: **open** and only return to real state after actual input event (change), so need to open and close gates before it comes normal.
I have made some loggings:

**First opening**
```
3064698144 shelly_main.cpp:443     Up 3064.69, HAP 0/1/12 ns 1, RAM: 28776/7896; st 0; 5.1: cur:open tgt:open cl:0 op:-1
3066529752 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3067260923 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3068261008 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3068516856 mg_rpc.c:310            Shelly.SetState via WS_in 192.168.88.125:62260
3068523143 shelly_output.cpp:63    Output 1: off -> on (GDO:close)
3068530094 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (RPC)
3068536802 shelly_hap_garage_d:222 GDO 1: Cur State: open -> closing (0 -> 3)
3068561880 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3069029458 shelly_output.cpp:63    Output 1: on -> off (pulse_off)
3069262061 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3070267652 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3071268294 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3072275214 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3072698697 shelly_main.cpp:443     Up 3072.69, HAP 0/1/12 ns 1, RAM: 28668/7896; st 0; 5.1: cur:closing tgt:closed cl:0 op:-1
3073272120 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3074280141 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3075273294 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3076270695 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3077268619 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3078273506 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3079270325 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3080268893 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3080698209 shelly_main.cpp:443     Up 3080.69, HAP 0/1/12 ns 1, RAM: 28668/7896; st 0; 5.1: cur:closing tgt:closed cl:0 op:-1
3081265508 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3082266894 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3083265315 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3084260325 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3085259016 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3086274343 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3087272799 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3088270469 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3088698400 shelly_main.cpp:443     Up 3088.69, HAP 0/1/12 ns 1, RAM: 28668/7896; st 0; 5.1: cur:closing tgt:closed cl:0 op:-1
3089267414 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3090265135 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3091261976 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3092260051 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3093259497 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3094274736 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3095275541 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3096271204 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3096698266 shelly_main.cpp:443     Up 3096.69, HAP 0/1/12 ns 1, RAM: 28668/7896; st 0; 5.1: cur:closing tgt:closed cl:0 op:-1
3097269742 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3098386354 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3099267116 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3100264338 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3101261594 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3102261660 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3103261557 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3104698309 shelly_main.cpp:443     Up 3104.69, HAP 0/1/12 ns 1, RAM: 28668/7896; st 0; 5.1: cur:closing tgt:closed cl:0 op:-1
3106414922 mgos_http_server.c:180  0x3fff51cc HTTP connection from 192.168.88.125:62973
3106427282 mg_rpc.c:310            Config.get via HTTP 192.168.88.125:62973
3108614676 shelly_hap_garage_d:222 GDO 1: Cur State: closing -> stopped (3 -> 4)
3112698291 shelly_main.cpp:443     Up 3112.69, HAP 0/1/12 ns 1, RAM: 28484/7896; st 0; 5.1: cur:stopped tgt:closed cl:0 op:-1
```

**First closing:**
```
3145261447 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3145603903 mg_rpc.c:310            Shelly.SetState via WS_in 192.168.88.125:62260
[... cut ...]
3161256527 mg_rpc.c:310            Shelly.GetInfoExt via WS_in 192.168.88.125:62260
3168698119 shelly_main.cpp:443     Up 3168.69, HAP 0/1/12 ns 1, RAM: 28768/7896; st 0; 5.1: cur:opening tgt:open cl:0 op:-1
3176698087 shelly_main.cpp:443     Up 3176.69, HAP 0/1/12 ns 1, RAM: 28768/7896; st 0; 5.1: cur:opening tgt:open cl:0 op:-1
3180639043 shelly_input.cpp:76     Input 1: change (state 0)
3180676610 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (ext)
3180683745 shelly_hap_garage_d:222 GDO 1: Cur State: opening -> closed (2 -> 1)
```

d4rkmen, 2023-08-10:
Hi, many weeks this problem annoying me :) So I have deviced to do "the deeper look"
And what I can see so far:
https://github.com/mongoose-os-apps/shelly-homekit/blob/ceb8cb37d98cc8649fae69b3bdc45eb2866dabff/src/shelly_hap_garage_door_opener.cpp#L98
Init for current and target state made without paying attention to input sensor mode

markirb, 2023-08-31:
Judging by the code this is correct. Why don't you just fork, and add this, let GitHub actions automatically build it and then try it?

timoschilling, 2023-09-01:
Or simply provide a PR with that change

d4rkmen, 2023-12-11:
@timoschilling done

d4rkmen, 2024-08-28:
The problem was in my installation, not the FW

closed 2024-08-28 by d4rkmen