mongoose-os-apps/shelly-homekit #1025: Shelly Uni - garage door opener

Kuddelmuddel84, 2022-07-04:
hello, the shelly uni has 2 exits. when i use it as a garage door opener i can only take one exit.
Why is that ?
I would like to use both exits because I have a 2-leaf gate and both wings can be opened separately.

Thanks for reply.

Kuddelmuddel84, 2022-07-07:
Hallo, sprichst du zufällig deutsch ? Dein Name klingt so :-) 
Also wird man mit dem Shelly Uni im Garage Door Mode niemals beide Ausgänge nutzen können ? 
Mir würde ein Mangnetkontakt (Input) für beide Ausgänge genügen. 

Mit freundlichen Grüßen 

Manuel Kudlik

timoschilling, 2022-07-07:
Es werden beide Ausgänge genutzt wenn du ein Tor hast welches getrennte Schalter für auf und zu hat. 

Man könnte einen 2x Garage Modus einbauen aber der Aufwand ist nicht grade klein. 

Besorg dir einfach nen zweiten Uni und gut ist