mongoose-os-apps/shelly-homekit #1027: Cannot flash over 0.10.3

AlRam79, 2022-07-05:
I have two Shelly 1L that I cannot apply shelly-homekit to. Units are running stock firmware version 0.10.3, build ID 20220617-112651/0.10.3-g7c89a05. 
When trying OTA, the device seems to revert to stock firmware. I've also tried ./flash-shelly.py to no avail. When running ./flash-shelly.py -a, I get no upgradeable devices. I have no VLANS or similar, just plain home WiFi with two Ubiquity APs. 

Also, running ./flash-shelly.py --local-file= hostname results in "Host: 10.0.0.106 is not a Shelly device."

The interesting thing is that I flashed one of the devices a short while ago just to test it and it went well. When reverting to stock firmware, it won't go to homekit again. The other device won't accept homekit at all as it seems. 

Anyone else?

AlRam79, 2022-08-07:
Hasn't anyone else experienced this issue?

soujos, 2022-08-14:
I’m facing the same issues with Shelly 1 plus. The first time everything worked OK, then I had to revert back to the original firmware and I tried to flash again the HomeKit firmware I couldn’t do it anymore, the device gets frozen and I have to power off and power on again and it keeps always the original firmware. 
I’ve tried many different things and nothing worked, any thoughts about what the issue is?

qubabos, 2022-08-22:
have the same issue, reverted to 2.11 beta, then manual to 2.11.2 and next reverted to stock.

closed 2022-09-29 by github-actions[bot]