mongoose-os-apps/shelly-homekit #1029: Flashing lights

AStoker, 2022-07-10:
Just started happening, but the lights will flash off and on every 2 seconds or so. Seems to maybe have something to do with the Wi-Fi connection, as it will disconnect sometimes. Can't seem to find documentation as to why this is happening. Seems to happen randomly and will occasionally self resolve. But randomly flashing lights isn't great (especially while watching the latest season of Stranger Things).

rojer, 2022-07-10:
can't imagine why it would be happening... does the device reboot? check uptime in the web interface, does it reset or keep ticking?

AStoker, 2022-07-10:
I think it's rebooting frequently. The uptime is never that long. I'm about to try and revert to factory firmware to see if that solves it. It's not having this trouble on the other 3 Shelly devices running the same HomeKit firmware.

AStoker, 2022-07-10:
Interestingly, I turned on AP mode to try and connect directly to it better and whenever I connect to it the lights flash.

rojer, 2022-07-10:
this looks like hardware issue related to power stability.

markirb, 2022-07-21:
I had this experience now. It only occurs for me when the power-saving for wifi is enabled (1 = light sleep). Does not happen otherwise (disabled or 2 = modem sleep)

markirb, 2022-07-21:
Ok forget it, this is not power related:

See: https://blog.creations.de/?p=149

quote:
Because of the halted system clock some peripherals might not work in this mode, use Auto Modem Sleep in those cases.

rojer, 2022-08-22:
right, light sleep interferes with timers. don't do it.

closed 2022-08-22 by rojer