mongoose-os-apps/shelly-homekit #1037: HomeKit update threshold too low

jameswestcott, 2022-07-17:
I’m using Shelly Plus 1PM with firmware 2.11.1 and notice HomeKit is receiving energy updates at least every second. It seems the threshold to send an update is very low. Although it’s working I’m concerned as I scale up with more devices it’ll have a negative impact on the HomeKit platform.

markirb, 2022-07-17:
You're right it updates every second in worst case (value not steady). This is not only for Shelly 1 PM, but all Shellys with a Powermeter.
.
https://github.com/mongoose-os-apps/shelly-homekit/blob/master/src/shelly_switch.cpp#L395
https://github.com/mongoose-os-apps/shelly-homekit/blob/master/src/shelly_switch.cpp#L404
The notification is called in the second link.
I guess it would be best, if the notification interval is adjustable in the web interface?

jameswestcott, 2022-07-17:
Yes a control in the web interface would do it. Either to adjust the interval or to adjust the change threshold for when to send an update.

thatslolo, 2022-07-18:
The second proposal would be nice!

rojer, 2022-07-26:
i think we can increase the interval to 5 seconds with no significant usability impact.
we should also fuzz it a little, in case multiple devices are restarted at the same time (like after power outage), we don't want them to report all at once.

rojer, 2022-07-26:
done. it will be reporting at 5+/-0.5 seconds interval now.

rojer, 2022-07-26:
i don't think it's worth exposing in the interface, 5 seconds should be enough for everyone (c) :)

closed 2022-07-26 by rojer