mongoose-os-apps/shelly-homekit #1039: Shelly2.5 Wiring garage door

louis49, 2022-07-17:
Hello,

I followed the wiki to wire my garage door : the Shelly25 is starting well but action is not working : garage door do not move.

Here my wiring diagram : 
[image]

When I directly connect 24v and Sbs, garage door move.

I understand that 24v and Sbs need to be behind a switch but how can I do it with shelly2.5 ? If I invert polarity, the shelly do not start but wiring seems good, when a set the good polarity the shelly start but not rely 24v and Sbs.

Thank you for your help

rionshin, 2022-07-17:
The movement of your door is happening from pin 4 - Stop. SBS is for on/off nothing else . Also Shelly 2.5 is for direct control of the motor of the door where motor is connected between Shelly out 1 and 2 - check the scheme here https://shelly.cloud/knowledge-base/devices/shelly-25/
You can still use the Shelly 2.5 with controlling out 1 to open the door and stop function can be connected to out 2 but you need resistor . So you will have 2 buttons exposed from Shelly 2.5 - one for SBS your switch 1 and one for stop your switch 2.  In your original scheme replace switches with Sw1 of Shelly and Sw 2 of Shelly. For switch 2 it’s written that you need 82k resistor .

louis49, 2022-07-19:
Thank you for your explanations.

Here the modified wiring diagram by adding a dry contact relay : 
[image]

Is that ok ?

closed 2022-08-26 by github-actions[bot]